libretro/Lakka-LibreELEC #1354: lakka 3.0 Chinese display problem

root-pi, 2021-05-31:
I installed version 3.0 of Lakka on the Raspberry Pi 3B+ and selected Chinese as the language, but the Chinese display was incomplete.Some words show squares.The more fatal problem was that after I rebooted the system.They all appear as squares and don't show any characters.Could you please fix the Chinese display error when updating the version?Thanks a million!

gouchi, 2021-05-31:
Hi,

Did you change the font in between your actions ?

As it seems it causes an [issue](https://github.com/libretro/RetroArch/issues/12215).

root-pi, 2021-06-01:
> 你好，
> 
> 它可能与这个[问题有关](https://github.com/libretro/RetroArch/issues/12215)？

Thanks for your reply.Yes, I'm using Lakka 3.0 exactly like this one.X MB theme.The Chinese display problem remains unresolved.Please also fix this problem when you update the version.Thanks again!

ToKe79, 2021-06-18:
can you please test with [latest nightly build](https://nightly.builds.lakka.tv/latest/)?

atlus666, 2021-06-23:
hi , this problem can be fix
because the theme default font not support chinese 
you can upload any chinese font (ex: TaipeiSansTCBeta-Bold.ttf) to your lakka
first change the theme font to your font 
and change system language you will see the complete chinese menu

root-pi, 2021-06-25:
> 你能用[最新的每晚构建](https://nightly.builds.lakka.tv/latest/)测试吗？

I'm sorry to see that the problem I tested in version 3.2 still exists, and the system font can be displayed normally after replacing it with Yahei, but after rebooting it, it all becomes a block and does not display any characters.

root-pi, 2021-06-25:
> hi , this problem can be fix
> because the theme default font not support chinese
> you can upload any chinese font (ex: TaipeiSansTCBeta-Bold.ttf) to your lakka
> first change the theme font to your font
> and change system language you will see the complete chinese menu

I have tried, is replaced by other font can be completely displayed.But if you reboot all the characters become squares.No substantive solution to the problem

Zalica, 2021-06-30:
I'm Chinese user and I have similar problem. 
In lakka 3.2, the rgui can display chinese font,althought it's not quite a pleasure sight. 
The xmb  glui  an ozone just can't display Chinese font by default.

root-pi, 2021-07-01:
是的。现在没有改过来。按照网上教程替换字体可以正常显示中文了。但是重启后显示的都是一个个小方块，一个中文都没有。

gouchi, 2021-07-11:
I did bisect and the first bad commit is [[8132303d4f9400e0a37051a4f7b43e26c6eb958e](https://github.com/libretro/RetroArch/commit/8132303d4f9400e0a37051a4f7b43e26c6eb958e)] (FontRenderer/FreeType) Use fontconfig to select fonts if available.

ToKe79, 2021-07-15:
Possible workaround for XMB menu is to download a font that has all the required glyphs and store it somewhere in the /storage partition. Then in RetroArch navigate to `Settings` -> `User Interface` - `Appearance` -> `Menu Font` -> locate the font you want to use. From this point that font will be used even after reboot/restart. Maybe changing the theme will reset to theme font, so this steps have to be reproduced again to display the characters correctly again.
Note: If you do not see Menu Font item, activate `Show Advanced Settings` in `Settings` -> `User Interface`.

linlinio, 2021-07-28:
I use the l4t 3.2 (Nintendo switch) version.i guess that lakka loads another ttfs in system starting.so i find to change files which locate at ”\usr\share\fonts\liberation\LiberationMono-Bold.ttf" and "\usr\share\fonts\liberation\LiberationMono-regular.ttf".and also replace ttfs under ozone.Then issue is resolved.

PR #1485 (2021-09-05): fontconfig: Add RetroArch fonts to font path
Fix #1354 #1362
closed 2021-09-05 by Ntemis