libretro/Lakka-LibreELEC #1358: [BUG] Computer Module 4 & Lakka 3.0

kelroy1990, 2021-05-31:
Hi guys, i am testing the lakka 3.0 release on the Computer Module 4 and the raspberry pi IO board.

The display and the HDMI its working well as i see the image correctly. But we have a few issue with the USB.

Computer Module 4 has his USB off by default, so we need to use the **dtoverlay=dwc2,dr_mode=host** to activate it. But on Lakka 3.0 it seems to dont have any kind of effect, so i cannot control the SO.

I see that Lakka have the dwc2 overlays on the /boot/overlays.... so i cannot figure where the problem is.

Its somebody testing it like me?.

Next Steps:
-Create a wpa_supplicant.conf on boot to try to obtain WiFi and debug by SSH.

Regards, Carlos.

closed 2021-07-23 by Ntemis

shawhall, 2021-07-25:
Did this get fixed?  I see the issue was closed but no comments were left.

Ntemis, 2021-07-25:
You didnt update the issue after all the stable releases and no one on the team has the Cm4 so no idea if it was fixed or not but we cant walk blindly. Try v3.2 and let me know though.

shawhall, 2021-07-26:
Just tried v3.2 today and it does not work.

kelroy1990, 2021-07-29:
Hi guys, tried to and it does not work. If the lakka team need one, i could send it to they, just contact with me.

I was cheking the kernel build and the dwc2 module seems to be active to compile, i need further work to see where its the fail.

dmrlawson, 2021-09-16:
Hi, I think this should work now in the latest nightlies if you enable it in distroconfig.txt