libretro/Lakka-LibreELEC #1361: SMB access issues

Orofire, 2021-06-01:
Hello,

I've just installed Lakka on a RPI 4.
I am now trying to access it through SMB but without success.
The service is running on Lakka.
The box is well listening on port TCP 445 (checked using netstat).
The telnet from 2 different computers is working fine but cannot access it using WIndows Explorer.
I've also tried to activate CIFS/SMB1.0 support within windows services. 

Is there something I'm missing ? 

Thanks for you help!

Orofire, 2021-06-01:
update : using Lakka 3.0 and Windows 10 (20H2)
disabled Widnows firewall + antivirus to test

gouchi, 2021-06-01:
Hi,

Can you try to make a test by adapting the [version](http://lakka.tv/doc/Serving-ROMs-from-a-NAS/) parameter  according to the [Samba version](https://docs.microsoft.com/en-us/windows-server/storage/file-server/troubleshoot/detect-enable-and-disable-smbv1-v2-v3) you are using on Win10 ?

Orofire, 2021-06-01:
Hi Gouchi thanks for the reply.
However, I am not sure how to proceed as the filesystem is in read-only mode.
I've tried to mount it in RW to be able to edit the smb.conf file but did not succeed.
Here is what command I used : mount -o remount,rw /flash

Just for the records, did you experienced the same issue with your config ? (if you also use LAKKA 3.0 ver)

Thanks!

gouchi, 2021-06-01:
Hi,

I am sorry I made a mistake. You are trying to access it with the smb share from your RPi so it is the contrary of the documentation.

Did you try to access it [directly with the IP](http://lakka.tv/doc/Accessing-Lakka-filesystem/#file-transfer-via-samba) ?

Thank you.

Orofire, 2021-06-01:
Ha yes indeed :) 

So, just to make it clearer : 
I'm just trying to access LAKKA from my Windows 10 computer using Windows Explorer.
So it uses SMB protocol.

Yes, I've tried via the IP.
Ping is working.
SSH is working.
Telnet from windows machine to LAKKA box on port TCP 445 is working.

I've also checked if the box is well listening on port TCP 445 and seems to be OK too :
tcp        0      0 0.0.0.0:139             0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:111             0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:445             0.0.0.0:*               LISTEN
tcp        0    448 192.168.1.13:22         192.168.1.95:40672      ESTABLISHED
tcp        0      0 :::139                  :::*                    LISTEN
tcp        0      0 :::111                  :::*                    LISTEN
tcp        0      0 :::22                   :::*                    LISTEN
tcp        0      0 :::445                  :::*                    LISTEN

So my last guess is indeed the SMB version that is too old to be allowed by Windows 10 but even if I enable SMB 1.0/CIFS, it is not working.
Really strange.
Did you see other people reporting this issue on LAKKA 3.0?

Thanks,

ToKe79, 2021-06-02:
I know this might seem obvious, but I rather ask - have you enabled the samba service on lakka?

Orofire, 2021-06-02:
Hello Toke79,

Indeed, I even checked if the service is well running using : systemctl status smbd
Service is running.
I just some a fresh re-install of Lakka without success.
I also tested using wired card instead of the wireless. 

Regarding the config file located in /storage, I already checked and there is a sample conf file for samba in /storage/.config but nothing else.
I can copy and edit it but I am not sure where to place is afterwards in order to make it used by the system.
I tried to place it next to the sample file but not seems to be used after a reboot.

Thanks for your support.

ToKe79, 2021-06-02:
you do not have to place it anywhere - any changes will be applied with the next restart of the service / after reboot. You just need to remove the .sample suffix from the file for it to be considered. At least that is what I recall :-)

closed 2021-06-02 by Orofire

ToKe79, 2021-06-02:
default user:pass should be root credentials.