libretro/Lakka-LibreELEC #1362: issue with korean language setting

vus7294, 2021-06-02:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using? 3.0 stable
- What system hardware are you using? nintendo switch
- What did you do? setting to korean language
- What did you expect to happen? setting normally
- What happened instead? reboot, all font crashes(except latin language) and force to set english

and that's the bottom line.

gouchi, 2021-06-02:
I reproduced the issue 

- Change language to Korean OK
- Restart RA or reboot KO it displays square.

gouchi, 2021-06-02:
Maybe related to this [issue](https://github.com/libretro/RetroArch/issues/12215). We may have issue with CJK language.

ToKe79, 2021-06-18:
can you please test with [latest nightly build](https://nightly.builds.lakka.tv/latest/)?

vus7294, 2021-06-18:
No. the Bug is still present in latest nightly build.

gouchi, 2021-07-11:
I did bisect and the first bad commit is [8132303d4f9400e0a37051a4f7b43e26c6eb958e] (FontRenderer/FreeType) Use fontconfig to select fonts if available.

ToKe79, 2021-07-15:
Please see for possible workaround: https://github.com/libretro/Lakka-LibreELEC/issues/1354#issuecomment-880622077

gouchi, 2021-09-05:
It will be fixed in Lakka 3.4

closed 2021-09-05 by gouchi