libretro/Lakka-LibreELEC #1368: (Nintendo Switch) When I reboot to hekate I get "repair usb" on Windows 10

pitrako, 2021-06-02:
- Lakka-Switch.aarch64-nightly-20210602-fd7440d
- Nintendo Switch
- Use reboot option that reboots to hekate
- Rebooting to hekate (It reboots ok to hekate but...) 
- When I start lakka again It reboots to hekate after the lakka logo automatically, second time I choose boot to lakka It starts correctly but also I get "repair usb" when that happens on Windows 10 but the microsd works ok.

pitrako, 2021-06-03:
Fixed in "Lakka-Switch.aarch64-nightly-20210603-9a1c5a4".

pitrako, 2021-06-05:
I noticed that I still get It randomly. Latest build I'm using is "Lakka-Switch.aarch64-nightly-20210605-b5dccf9".

plaidman, 2021-06-05:
This may have something to do with rebooting while the card is being written to. Make sure you are not writing a save game/state and you have closed any running content before you reboot to hekate. Additionally if you have a corrupted file on your card and you don't fix it, you may still get the message until you do fix it properly.

pitrako, 2021-06-05:
I rebooted without closing a game and the problem didn't happen and each time the message pop ups on windows I click on repair and It says It didn't find any problem.

plaidman, 2021-06-06:
"I rebooted without closing a game" yes that is exactly the problem. it doesn't always happen, just when it's trying to write to disk. you may see some .REC files (I think that's the extension but I may be wrong) on your sd, which would tell you for sure.

pitrako, 2021-06-08:
I don't see any rec files on the microsd but it didn't happen anymore, I suppose you are right and was because of sometimes rebooting without closing content, I'm gonna close the issue, thanks.

closed 2021-06-08 by pitrako