libretro/Lakka-LibreELEC #1369: Wifi issue Orange Pi Lite

diegohvanni, 2021-06-03:
Wifi issue with Lakka V3.0 on an Orange Pi Lite (All winner H3) causes the wifi to not find any network and trying to turn the wifi off and on does not seem to make changes.

ToKe79, 2021-06-03:
Does the Wi-Fi work with LE 10 beta? (https://test.libreelec.tv/)

diegohvanni, 2021-06-03:
There's no version specfically for the lite pi (it does have for lite 2 tho). Whixh one should I use instead?

ToKe79, 2021-06-03:
Unfortunately there is no official support from LibreELEC for your device in current master/10-beta I guess - and there never was official support for Allwinner in 9.2 to my knowledge, so... support of Allwinner devices is just experimental. I'm not sure if any of the devs/contributors owns an Allwinner board to test, so I doubt this will change unless there is support for the device from upstream (= LibreELEC).

diegohvanni, 2021-06-03:
Gotcha. I may help testing it myself. I don't  know if its poasible to customize the existing rom to add the wifi support. However, it used to work very well on the previous version, so I would try to "port" from one to another. I will give a better look later today and share my findings here.

Ntemis, 2021-06-06:
Not supported by mainline kernel.

closed 2021-06-06 by Ntemis