libretro/Lakka-LibreELEC #1372: Adding the PCSX2 core

Abdul-Muiz-Iqbal, 2021-06-05:
Can the PCSX2 core be added into Lakka, or is there some problem which prevents it from being usable?

ToKe79, 2021-06-07:
The core does build only for Windows. Also it requires x86 64-bit processor, so only the `Generic 64-bit PC` platform would benefit from it. Sorry, but currently not doable to include this core in Lakka, as it is based on Linux. To use PCSX2 you have to use RetroArch on Windows or the standalone/upstream emulator.

closed 2021-06-07 by ToKe79

dokoma, 2022-03-25:
Hello @ToKe79 ! I tried again and again on my generic-pc-x64-nvdiaGTX1050 with pcsx2 core but with no luck and no any further log ,until i see this post.  it crashed and restart retroarch everytime, Is it still "windows-x86-64-only-core" today?  btw:  I tried batorcera 3.3 with the same pc, besides its standolone pcsx2, the pcsx2-core works, could that core be used?