libretro/Lakka-LibreELEC #1374: Feature request - Friendly access to TTY mode

WHK102, 2021-06-05:
Please, add an option to access to console or tty mode from lakka interface, by xample, in services menu.

davidgfnet, 2021-06-06:
Yeah this is pretty much needed.
There are some docs in the wifi and lakka help docs, did any of them work? I recall some might work (and you get a TTY on the device). UART tty is another story tho.

shantigilbert, 2021-06-07:
Doesn't it go to a terminal if you add the `textmode` in the boot env vars?

gouchi, 2021-06-08:
What is the use case ? As @shantigilbert's note, you can boot to the terminal with `textmode` boot parameters.

On which platform ?

ToKe79, 2021-06-08:
also note that one has to add `retroarch=0` to be dropped directly to TTY (instead of going to RetroArch), if there is no ssh access to stop the retroarch service. so adding `textmode retroarch=0` to the cmdline gives root prompt after the boot. there is currently no other possibility to start a TTY / open shell and also not planned to add any such additional feature.

closed 2021-06-08 by ToKe79

davidgfnet, 2021-06-08:
Is this in the docs somewhere? Would be great to add it otherwise! Thanks!

WHK102, 2021-06-09:
But you need edit boot options before start, require restart all if need run a simple command from lakka without ssh, and, the boot options is not friendly. By example, to get status of usb controller using a simple lsusb. In my case, the lakka machine is in the A room and my pc is in B room, need walk between two rooms for each command o movement in to controller?, is better option connect usb keyboard to machine and test minimal commands when game is run.

gouchi, 2021-06-09:
@davidgfnet We have some docs [here](http://lakka.tv/doc/Accessing-Lakka-command-line-interface/#direct-access).

@WHK102 There are two requests : for the cheat please check the [documentation](https://docs.libretro.com/guides/cheat-codes/). For the second one, I am not sure at all it will be be integrated to RA you should open a [feature request on RA issue tracker](https://github.com/libretro/RetroArch/issues). On Lakka, for now you will have to submit boot parameters or connect by SSH. We could try to enable [createlog](https://github.com/LibreELEC/LibreELEC.tv/search?utf8=%E2%9C%93&q=createlog&type=) to samba share also.

WHK102, 2021-06-09:
ok, to connect to the terminal from the same machine it is necessary to have two monitors and two operating systems unless it is possible to connect by ssh to localhost just to have a shell, it is something absurd.