libretro/Lakka-LibreELEC #1377: NVME SSD M.2 hard drive is not recognized in Lakka 3.1

matcarfer, 2021-06-07:
- Which version of Lakka are you using? 3.1
- What system hardware are you using? Asus H110M-E/M.2 + Pentium G4560 + Kingston A2000 240GB NVME SSD M.2
- What did you do? Tried to access my ROMS on NVME SSD.
- What did you expect to happen? To work.
- What happened instead? It didn't.

Attached 2 logs

[blkid.txt](https://github.com/libretro/Lakka-LibreELEC/files/6605343/blkid.txt)
[dmesg_mount.txt](https://github.com/libretro/Lakka-LibreELEC/files/6605344/dmesg_mount.txt)

This draw my attention:
[    0.610605] nvme nvme0: pci function 0000:01:00.0
[    0.610624] ahci 0000:00:17.0: version 3.0
[    0.610920] ahci 0000:00:17.0: AHCI 0001.0301 32 slots 4 ports 6 Gbps 0xf impl SATA mode
[    0.610922] ahci 0000:00:17.0: flags: 64bit ncq sntf led clo only pio slum part ems deso sadm sds apst
[    0.616871] nvme nvme0: missing or invalid SUBNQN field.
[    0.618123] nvme nvme0: 4/0/0 default/read/poll queues

gouchi, 2021-06-08:
Hi,

Thank you for the log.

It is strange as your partitions are well detected.

If you try to mount it manually with `mkdir win && mount -t ntfs /dev/nvme0n1p4 win`, what do you get ?

Do you have fast boot/startup or encryption enabled on Windows ?

Ntemis, 2021-07-25:
Issue looks stalled. Closing up.

matcarfer, 2021-07-26:
Sorry for taking too long to respond. Maybe we can open the issue again?
@gouchi that doesnt do nothing as far as I can see, no error and no message.
Cant see the storage. Attached newer logs from latest nightly (Lakka-Generic.x86_64-nightly-20210724-58dafbe.img).
Also I made sure I deactivated fast startup on win10 to avoid issues (it wasnt before) and I dont have any encryption on the drive.
[dmesg_mount.txt](https://github.com/libretro/Lakka-LibreELEC/files/6875253/dmesg_mount.txt)
[blkid.txt](https://github.com/libretro/Lakka-LibreELEC/files/6875254/blkid.txt)

closed 2021-07-27 by Ntemis

matcarfer, 2021-08-02:
Can confirm that its now fixed on Lakka 3.3. This is great! Thanks!