libretro/Lakka-LibreELEC #1381: Feature request - Contain dm-mod.ko by default.

ventoy, 2021-06-08:
Hi,
With [Ventoy](https://github.com/ventoy/Ventoy) we can just copy the lakka.img to the USB drive and boot it.
But Ventoy need the `dm-mod.ko` which is not included in the img file.
Can we contain the `dm-mod.ko` in the img file by default in the future release?

Thanks!

gouchi, 2021-06-08:
Hi,

It seems it will require to enable device mapper [CONFIG_BLK_DEV_DM](https://cateee.net/lkddb/web-lkddb/BLK_DEV_DM.html).

Which platform do you support with ventoy ? According to your [website](https://www.ventoy.net/),

> x86 Legacy BIOS, IA32 UEFI, x86_64 UEFI, ARM64 UEFI and MIPS64EL UEFI

So we will have to enable it for i386, x86_64 and aarch64 platform ?

Thank you.

ventoy, 2021-06-08:
Yes.
Currently,  `CONFIG_BLK_DEV_DM` is not enabled in the dot.config.
To support lakka 2.3.2/3.0/3.1 I built the linux kernel with your dot.config(with CONFIG_BLK_DEV_DM=m) and include the
dm-mod.ko in Ventoy package. So if we contains the ko in the img file in the future, it will be great.

For lakka `i386, x86_64, and aarch64` are supported.

Thanks.

ventoy, 2021-06-08:
Since the ko files are in the squashfs image file `system`. 
So it's better to set `CONFIG_BLK_DEV_DM=y` to build the module directly into the kernel. Then I will no need to extract the ko file before I boot the kernel in the bootloader period.

Thanks.

gouchi, 2021-06-12:
Can you make a test with this [image](https://nightly.builds.lakka.tv/members/gouchi/1381/Lakka-Generic.x86_64-3.0-devel-20210612151625-9a78362.img.gz) ?

Thank you.

ventoy, 2021-06-12:
Perfect!
The `dm-mod` is enabled by default and Ventoy can boot it fine.

Thank you.

ToKe79, 2021-06-14:
we keep this open until @gouchi submits the fix :-)

PR #1395 (2021-06-17): linux: Enable CONFIG_BLK_DEV_DM on Generic
- Fix #1381
closed 2021-06-17 by ToKe79