libretro/Lakka-LibreELEC #1383: GPiCase with Pi Zero W: Beetle Supafaust not successfully launching.

yikesmeguts, 2021-06-09:
*Which version of Lakka are you using?* Lakka-GPICase.arm-3.0-devel-20210608233405-7af0ecf
*What system hardware are you using?* RetroFlag GPiCase with Raspberry Pi Zero W, new microSD card
*What did you do?* Installed this particular PR build to see if the changes have corrected the recent issues with the rotated display in RetroArch and with crashing due to WiFi connections have been corrected.  Tested a variety of system cores, mostly of 8-bit and 16-bit system to see how they perform in comparison to Lakka 2.3.2 for GPiCase.
*What did you expect to happen?* I expected to find some regressions as well as improvements as part of my testing.
*What happened instead?* I found that most of the cores I tried launched properly and ran very smoothly.  However, I found that if I selected Beetle-Supafaust as the core for any SNES rom, the core would not launch, returning to RetroArch shortly after attempting to launch Supafaust.  Due to the small screen, I was unable to read the tiny message that popped up upon returning to RetroArch.

gouchi, 2021-07-07:
As stated in issue template, please provide some log.

> It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at https://www.lakka.tv/doc/Troubleshooting-Lakka/

Thank you.

yikesmeguts, 2021-07-07:
Apologies for not including the log when the issue was submitted. I will submit within 24 hours. 

        --CW

----
Christopher L. Wright
Information Technology Consultant
College of Visual and Performing Arts
University of North Carolina at Greensboro
***@***.***

yikesmeguts, 2021-07-07:
Included below is the logfile generated by the createlog command.  The device is a GPiCase with a Pi Zero W running Lakka 3.2.  I set the core association for the 'snes' roms directory to Beetle Supafaust, then launched Super Mario World.  After a few seconds, the system returned to the Retroarch screen with a popup message saying "failed to load content".  I then set the core association to Snes9x 2002 and launched Super Mario World again.  Snes9x 2002 loaded the rom successfully.  At this point, I restarted retroarch, copied the logfile to my workstation, and attached it to this message to send to you.

I hope this helps.  I have been impressed with the performance of Beetle Supafaust on my Pi 3B+ and Pi 4.  If the Pi Zero has enough horsepower to run Supafaust, it would make a nice addition to Lakka.  As it stands, Snes9x 2002 has thus far been satisfactory to meet my somewhat minimal SNES gaming needs.
Thanks to you and to the whole Lakka development team for making such a fabulous product.

yikesmeguts, 2021-07-07:
[log-2021-07-07-19.21.56.zip](https://github.com/libretro/Lakka-LibreELEC/files/6779705/log-2021-07-07-19.21.56.zip)

gouchi, 2021-07-09:
It is strange as we don't see any crash in the log.

May you try to check if you still reproduce the issue with [nightly builds](https://nightly.builds.lakka.tv/latest/GPICase.arm/) ?

Thank you.

yikesmeguts, 2021-07-11:
I have performed a clean install of the 11 July 2021 nightly on my GPi Case. Using createlog, I first opened Super Mario World with Snes9x 2002, which opened properly.  I then exited Snes9x, set the core association to Beetle Supafaust, then attempted to open Super Mario World again.  After several seconds, the Retroarch screen reloaded, displaying the message, "Failed to load content".  

[log-2021-07-11-01.39.33.zip](https://github.com/libretro/Lakka-LibreELEC/files/6796208/log-2021-07-11-01.39.33.zip)

gouchi, 2021-07-11:
Thank you for the log.

I see that there was a [new commit ](https://github.com/libretro/supafaust/commit/e4b69082a374ef788884543a5c22173cb91adcff)about adding RPi plaform to the Makefile which could help.

gouchi, 2021-07-14:
@yikesmeguts Can you make a test with this [nightly build ](https://nightly.builds.lakka.tv/PRs/pr_1426_000393/GPICase.arm/Lakka-GPICase.arm-3.0-devel-20210713191800-05406b0.img.gz)(RA 1.9.6 + updated supafaust core) ?

Thank you.

yikesmeguts, 2021-07-14:
@gouchi Yes, I will install this nightly on my device tonight and I will report my findings here. Thanks.

yikesmeguts, 2021-07-14:
@gouchi I downloaded the Lakka-GPICase.arm-3.0-devel-20210713191800-05406b0 build that you specified, and I installed it on a fresh microSD card in my GPi Case.  After transferring over a folder of SNES roms, I created an SNES playlist and selected Beetle Supafaust as the preferred core.  Upon attempting to launch several titles, each returned to the Retroarch menu with the message, "Failed to load content".  I then turned on logging via SSH and createlog, then attempted to launch 3 titles: Donkey Kong Country 2, Kirby's Dream Land 3 and Super Mario Kart. each failed to load content.  I then launched each with Snes9x 2002 successfully.

For the record, I don't play very many SNES games on the GPi, so Snes9x is sufficient to meet my needs.  When the first Lakka 3.x PR build was released which corrected the screen orientation issue, I tested it out and that was when I noticed the problem with Supafaust.  When I mentioned it on Discord, it was suggested that an issue be created on Github, which is what led to this particular issue.  I did not submit the issue out of frustration or dissatisfaction, but rather in the hopes that a potential problem could be identified and hopefully resolved for the benefit of the Pi zero/GPi case community of users.  I am more than willing to assist with testing as long as you need users to help test.  As always, thank you for the time and effort that you and the team continue to put into this project.
[log-2021-07-14-20.00.33.zip](https://github.com/libretro/Lakka-LibreELEC/files/6818696/log-2021-07-14-20.00.33.zip)

jdalmanza, 2021-09-29:
i think by default supafaust is designed for multi-core CPUs and IIRC pi Zero is only a single core. Is it possible that the build is expecting multiple cores and crashing because there is only a single? 

two important things I took away from the project page:

    _Minimum recommended CPU is a dual-core CPU, running at 900MHz for most SNES games, and at 1200MHz for SuperFX, SA-1, and CX4 games.
    ...
    Setting "supafaust_renderer" to "st", to disable multithreaded PPU rendering, may improve performance on systems with only a single CPU with a single core._

https://github.com/libretro/supafaust

yikesmeguts, 2021-09-29:
I am not heavily invested in trying to run SNES games on the GPi case, I usually reserve this device for older and lower-end game consoles that fall within the limits of what the Pi Zero can handle. When I do feel like playing SNES games, I either play them on a Pi4 attached to a TV screen, or on my OGS, both of which handle Beetle Supafaust fine.

When the Lakka 3.x code tree was adjusted to restore the screen orientation on the GPi case, I mention this issue with Beetle Supafaust on the Lakka Discord channel, and it was suggested that I open an issue on GitHub to make it official. 

I am satisfied that the Pi Zero will likely never be able to run this particular core. I have alternatives available that meet my needs, so I have no objection to closing this issue

Thank you and all of the Lakka team for all the work you put into this fine project.

gouchi, 2021-09-29:
@jdalmanza Thank you about this information I missed it.

@yikesmeguts you can try to add core override in `/storage/.config/retroarch/config/Supafaust/Supafaust.cfg`

```
supafaust_audio_rate = "disabled"
supafaust_correct_aspect = "enabled"
supafaust_cx4_clock_rate = "100"
supafaust_deinterlacer = "bob_offset"
supafaust_frame_begin_vblank = "enabled"
supafaust_h_filter = "phr256blend_auto512"
supafaust_multitap = "disabled"
supafaust_pixel_format = "rgb565"
supafaust_region = "auto"
supafaust_renderer = "st"
supafaust_run_ahead = "disabled"
supafaust_slend = "223"
supafaust_slendp = "238"
supafaust_slstart = "0"
supafaust_slstartp = "0"
supafaust_superfx_clock_rate = "100"
supafaust_superfx_icache = "disabled"
supafaust_thread_affinity_emu = "0x2"
supafaust_thread_affinity_ppu = "0x1"
```

For your case the most important is `supafaust_renderer = "st"`

jdalmanza, 2021-09-30:
i did a fresh install of 3.4 and tried to launch a snes game with the supafaust core and it crashed. I cannot verify it says the same message as the on-screen notifications are too small (see: https://github.com/libretro/Lakka-LibreELEC/issues/1448) but when I went to the location of the config file you mentioned above there is no file there. I'm not sure if I can copy/paste that into a new file, i'll try that next.

yikesmeguts, 2021-09-30:
I was planning to try this as well once I get home from work tonight.

jdalmanza, 2021-10-01:
copying the values into a new .cfg file did not work. Core still crashes

gouchi, 2021-10-02:
Thank you for tests.

May you try to get some log ? If it is a segfault we should try to open an [issue](https://github.com/libretro/supafaust/issues)  in the core repository with debug information.

Thank you.

yikesmeguts, 2021-11-14:
Update:  I just updated my GPiCase/Pi Zero 1 W to Lakka 3.6 and I was able to open multiple NES roms using Beetle Supafaust.  I don't play many SNES titles on my GPiCase, and I have been able to get by with the older Snes9x core, so it has been a while since I have tried Beetle Supafaust.  When I opened this issue a few months ago, Supafaust would not launch for me at all.  It is working for me now, however.  I'm not sure what changed (or when) to make things work again, but I certainly want to thank everyone who helped test and troubleshoot this matter, and I would especially like to thank those who fixed the problem.

jdalmanza, 2021-11-15:
interesting @yikesmeguts i just upgraded mine and it still seems to crash. I verified that the config override is present for the "st" single-threaded option but still did not work. I'll try a fresh flash on a new SD and see if the behavior is the same

yikesmeguts, 2021-11-15:
Further update: it was definitely not my imagination, I was able to launch several random roms in Supafaust, switched core association to Snes9x, and then switched back to Supafaust. On the PiZero1, Snes9x definitely performed more smoothly, Supafaust was a bit choppy with significant audio stuttering. Some roms would not launch in Supafaust, but others definitely would. When I opened the issue, Supafaust would not launch any rom that I tried. 

Upon successfully launching a rom (Kirby Super Stars) I changed one setting, I wnabled frameskipping to see if it would make a noticeable difference, and upon closing and relaunching Supafaust, it will no longer launch, even roms that had successfully launched prior to changing that one setting.

Now that the Pi Zero 2 is available and early test builds are available, I’m even less bothered by this. But I know there are many, many GPiCase users with Pi Zero 1’s that may not be interested in upgrading.

ToKe79, 2022-04-01:
@yikesmeguts @jdalmanza thank you for your feedback. I'm closing this old stale issue. Feel free to open new one in case there is anything we can do.

closed 2022-04-01 by ToKe79

yikesmeguts, 2022-04-01:
I agree that this issue has gone stale, and thanks to all who helped investigate the matter.