libretro/Lakka-LibreELEC #1387: Lakka OS freezes randomly while navigating (RPi 4) (Stable 3.1)

Rstation4, 2021-06-10:
Raspberry Pi 4 2gb
eeprom bootloader: Sep 3 2020
Micro Sd: 128 gb, Samsung Evo Select
Lakka OS version: Lakka-RPi4.aarch64-3.1

It has randomly freezes twice while I was just navigating through my playlists. 
my build details: Thumbnails for playlists, using Retrosystem theme.
I recreated it a third time by navigating through my playlists quickly, I have a log for it. It may not have logged the error but here is the log. 
[retroarch.log](https://github.com/libretro/Lakka-LibreELEC/files/6633198/retroarch.log)

ToKe79, 2021-06-10:
My guess this is not a bug, this could be simple performance issue (SD card I/O, image processing, ...). It would be good to have a comparable test (no log, just subjective experience with same setup in terms of SD card and other hw) with RaspberryPi OS + Retroarch started in linux console without display manager running (set the input driver to `udev` or `linuxraw` if you want to use keyboard before starting from console, as your keyboard might not "work" in RA, as I guess default will be `x`).

jayare5, 2021-06-11:
That's interesting... my Pi4 also froze for a bit at one point I was navigating a folder that was in a USB drive (MicroSD adapter). it took like 5 seconds to unfreeze and show the contents of the folder. I thought it was weird because I'm always navigating folders on the same device for years on older versions of Lakka and especially Pi3, never saw that happen before.

I also had it get completely frozen while I left it in the menu doing nothing for like 30 or so minutes while I went to do other stuff, when I came back it was frozen.

Initially I thought maybe it's because I was overclocking the Pi4 and didn't think it was too serious, but I left it for 30 or so minutes again and it didnt freeze. That's as far as my tests went.

Rstation4, 2021-06-15:
What would be the preferred specifics of a micro sd card for the RPi 4 running Lakka OS? For the [card](https://www.samsung.com/us/computing/memory-storage/memory-cards/evo-select-microsdxc-memory-card-128gb-mb-me128ha-am/) im using it has the following

Speed Class: Grade 3, Class 10
Read : up to 100MB/s with UHS-1 interface
Write : up to 60MB/s with UHS-1 interface*

will update when I can when I have another card to test further with

ToKe79, 2021-06-15:
It is the latency of microSD cards at random access. Sure, they can read/write in sequence quite fast, but the difference is that they lack a proper controller, so random access / parallel read/write performance is not comparable with hard drives. Also they lack any cache. I would maybe suggest performing a test with USB drive (not a pendrive, but regular hdd/ssd) and if there are still some freezes, it could be an issue not related to I/O performance. Also maybe https://github.com/libretro/RetroArch/issues/6747 is related to this issue?

Rstation4, 2021-06-15:
Does having a bunch of thumbnails add up to be a performance hit? I have 1,670  (564.2 Mbs used) thumbnails, not all thumbnails are in use. In prior versions on the Raspberry Pi 4 I did notice that bigger images around 2.5 Mbs tend to have a delay in appearance. Does having more playlists,  and with a larger number of playlist entries that have thumbnails slow down the UI?

Rstation4, 2021-06-22:
Did a clean install of **Lakka 3.2 aarch64** stable release on a brand new _128 gb SanDisk Ultra microSDXC UHS-I_ to see that it happend again while I was navigating the xmb menu. [Micro SD card specs](https://kb.sandisk.com/app/answers/detail/a_id/22697/)

Rstation4, 2021-07-03:
I did another test on a separate Raspberry Pi device, A **Raspberry Pi 4 (4 gb of Ram version)** and with a new[ PNY Elite 32gb micro sd card](https://www.pny.com/Elite-Class-10-U1-microSD-Flash-Memory-Card?sku=P-SDU32GU185GW-GE) witht the following. 
 1. A 3.2 aarch64 Rpi 4 stable Lakka image used
 2. Only had 3 playlists and a few games (29 roms in total)
 3. Had thumbnails added to playlist entries
 4. Audio set to hdmi
 5. Monochrome theme used, disabled animated background (I did that with the other device) and this time a custom background used.
 I navigated fast through the list by holding the d pad on up and sometimes cycling to the next playlist doing the same thing. Yet nothing so far. 
 
 To clarify the other devices prior were also new and had new sd micro cards. I have seen comments from others saying that the **4 gb of ram** is overkill for emulation on the **Raspberry Pi 4** is there any truth to that because what I'm seeing is contradicting it?

dmrlawson, 2021-07-06:
> I did another test on a separate Raspberry Pi device, A **Raspberry Pi 4 (4 gb of Ram version)** and with a new[ PNY Elite 32gb micro sd [...]

Are you saying this only happens with the 2GB rpi4?

With RAM the more the better, though this appears to not be an emulation bug but a Retroarch bug.

It's already been asked but could you check https://github.com/libretro/RetroArch/issues/6747 to see if it's the same issue?

Does it occur on menus other than XMB?

Rstation4, 2021-07-09:
@dmrlawson It appears to be the case with the 2 GB RAM versions. Though another thing I noticed is that Lakka OS stops and lags at the Import Content xmb menu when you navigate on both the 2 gb and 4 gb version of the RPi 4B. So far I had not run into any other problems with a 4gb Raspberry Pi 4 running Lakka 3.2 stable. On my current setup I changed things around to having my roms on a M2 ssd drive, the dynamic wallpaer off, animated wallpaper off to keep things with less performance hits. Even with my current build now having many playlists and thumbnails I don't see it freeze up.

2. As for your XMB question, In the previous stable versions  I had seen it crash on the Ozone GUI. I have not tested it yet with 3.2 for either the RPi 4 2gb and 4gb RAM versions.

closed 2021-07-23 by Ntemis