libretro/Lakka-LibreELEC #1388: Lakka 3.1 Mednafen psx hardware issue in video gl mode.

Carcharodon, 2021-06-11:
Good afternoon, I would like to report an error that is happening in Mednafen on ps1, when I try to use it in hardware mode as a video drive in the gl option, the emulator sends the message that it is not possible to display the hardware mode and that it will return to mode of software. This error occurs in the gl and glcore video driver, but it does not run in the Vulkan video driver.

ToKe79, 2021-06-11:
can you be more specific with your issue, e.g. which platform you use?

Carcharodon, 2021-06-11:
I am using 64bit linux pc version

Carcharodon, 2021-06-11:
I can use Medanfen on PS1 only in software mode or Vulkan, in hardware mode it has an error and the emulator automatically switches back to software mode

ToKe79, 2021-06-11:
what gpu? is vulkan not usable? are you able to provide [any logs](https://github.com/libretro/Lakka-LibreELEC/wiki/Troubleshooting-Lakka)?

closed 2021-07-25 by Ntemis