libretro/Lakka-LibreELEC #1389: Mupen64 plus next does not open in Lakka 3.1

Carcharodon, 2021-06-11:
Good afternoon, I would like to report an error that is happening in Mupen64 Plus Next, Lakka 3.1 cannot open any game through this emulator.

ToKe79, 2021-06-11:
when you open a new issue, the is a template - for a reason. can you be more specific with your issue, e.g. which platform you use?

Carcharodon, 2021-06-11:
I am using 64bit linux pc version every time I try to use Mupen64 Plus Next the game doesn't load and goes back to the Lakka start menu. This error only occurs with that emulator specifically.

ToKe79, 2021-06-11:
can you provide more details on your setup and some [logs](https://github.com/libretro/Lakka-LibreELEC/wiki/Troubleshooting-Lakka)?

Carcharodon, 2021-06-11:
I found the solution, the Mupen64 Plus Next emulator only works if the video driver is set to glcore or vulkan. Now I can play, the strange thing is that in other versions of Lakka there was no such problem.

closed 2021-06-12 by Carcharodon