libretro/Lakka-LibreELEC #1391: Vulkan driver not working with Dolphin on RPi4

sergimes, 2021-06-13:
Tried to play Super Mario Strikers (GameCube) with the Vulkan driver but after hitting run the screen stays on the flower logo screen and after some time it turns a little darker. Also tried with New Super Mario Bros Wii and the same happens. Note that the green led indicating SD writing is not blinking.
[image]

I'm using a RPi4 4gb and Lakka-RPi4.aarch64-nightly-20210611-9a78362, also tried on Lakka 3.1 and not working.

dmrlawson, 2021-06-13:
> Tried to play Super Mario Strikers (GameCube) with the Vulkan driver but after hitting run the screen stays on the flower logo screen and after some time it turns a little darker. Also tried with [...]

I seem to remember this working for me. 

Have you followed the setup instructions [here](https://docs.libretro.com/library/dolphin/)?

Could you provide logs from trying to run a game? (Instructions [here](https://www.lakka.tv/doc/Troubleshooting-Lakka/))

sergimes, 2021-06-14:
Done the setup instructions and still not working, here is the log:

###########################################
# Lakka - The DIY retro emulation console #
# ...... visit http://www.lakka.tv ...... #
###########################################

Lakka (official): devel-20210611013907-9a78362 (RPi4.aarch64)
Lakka:~ # systemctl stop retroarch.service ; retroarch -v
[INFO] RetroArch 1.9.4 (Git c226bd87f4)
[INFO] === Build =======================================
[INFO] Capabilities:  ASIMD
[INFO] Built: Jun 10 2021
[INFO] Version: 1.9.4
[INFO] Git: c226bd87f4
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[INFO] [Vulkan]: Found vulkan context: khr_display
[INFO] [Vulkan]: Detecting screen resolution 0x0.
[INFO] [Vulkan]: Found GPU at index 0: V3D 4.2
[INFO] [Vulkan]: Using GPU index 0.
[INFO] [Vulkan]: Found GPU at index 0: V3D 4.2
[INFO] [Vulkan]: Using GPU index 0.
[INFO] [Vulkan]: Using semaphores for WSI acquire.
[INFO] [Vulkan]: Using GPU: V3D 4.2
[INFO] [Vulkan]: Queue family 0 supports 1 sub-queues.
[INFO] [Vulkan]: Using resolution 1920x1080
[INFO] [Vulkan]: Using RGB565 format.
[INFO] [Vulkan]: GPU does not support using linear images as textures. Falling back to copy path.
[INFO] [Vulkan]: GPU does not support using linear images as textures. Falling back to copy path.
[INFO] [Vulkan]: GPU does not support using linear images as textures. Falling back to copy path.
[INFO] [Vulkan]: Loading stock shader.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[WARN] [udev]: Couldn't open any keyboard, mouse or touchpad. Are permissions set correctly for /dev/input/event*?
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [udev]: Pad #0 (/dev/input/event0) supports force feedback.
[INFO] [udev]: Pad #0 (/dev/input/event0) supports 16 force feedback effects.
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [Video]: Found display server: null
[WARN] [udev]: Mouse grab/ungrab feature unavailable.
[INFO] ALSA: Using floating point format.
[INFO] ALSA: Period size: 384 frames
[INFO] ALSA: Buffer size: 1536 frames
wifi is already enabled
[INFO] [Display]: Found display driver: "vulkan".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [SRAM]: SRAM will not be saved.
[INFO] [Playlist]: Loading history file: [/storage/.config/retroarch/content_history.lpl].
[INFO] [Playlist]: Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] [Playlist]: Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] [Playlist]: Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
[INFO] [Playlist]: Loading favorites file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] [Vulkan]: Destroying stale acquire semaphore.
[INFO] [Vulkan]: GPU does not support using linear images as textures. Falling back to copy path.
[INFO] [Vulkan]: GPU does not support using linear images as textures. Falling back to copy path.
[INFO] [Vulkan]: GPU does not support using linear images as textures. Falling back to copy path.
[INFO] [Vulkan]: QueuePresent failed, destroying swapchain.
[INFO] [Vulkan]: GPU does not support using linear images as textures. Falling back to copy path.
[... 692 lines cut ...]
[libretro INFO] 24:42:065 Common/FileUtil.cpp:209 I[COMMON]: CreateFullPath: path exists /storage/savefiles/User/Dump/Audio/
[libretro INFO] 24:42:065 Common/FileUtil.cpp:205 I[COMMON]: CreateFullPath: path /storage/savefiles/User/Dump/DSP/
[libretro INFO] 24:42:065 Common/FileUtil.cpp:209 I[COMMON]: CreateFullPath: path exists /storage/savefiles/User/Dump/DSP/
[libretro INFO] 24:42:065 Common/FileUtil.cpp:205 I[COMMON]: CreateFullPath: path /storage/savefiles/User/Dump/SSL/
[libretro INFO] 24:42:065 Common/FileUtil.cpp:209 I[COMMON]: CreateFullPath: path exists /storage/savefiles/User/Dump/SSL/
[libretro INFO] 24:42:065 Common/FileUtil.cpp:205 I[COMMON]: CreateFullPath: path /storage/savefiles/User/Dump/Frames/
[libretro INFO] 24:42:065 Common/FileUtil.cpp:209 I[COMMON]: CreateFullPath: path exists /storage/savefiles/User/Dump/Frames/
[libretro INFO] 24:42:065 Common/FileUtil.cpp:205 I[COMMON]: CreateFullPath: path /storage/savefiles/User/Dump/Objects/
[libretro INFO] 24:42:065 Common/FileUtil.cpp:209 I[COMMON]: CreateFullPath: path exists /storage/savefiles/User/Dump/Objects/
[libretro INFO] 24:42:065 Common/FileUtil.cpp:205 I[COMMON]: CreateFullPath: path /storage/savefiles/User/Dump/Textures/
[libretro INFO] 24:42:065 Common/FileUtil.cpp:209 I[COMMON]: CreateFullPath: path exists /storage/savefiles/User/Dump/Textures/
[libretro INFO] 24:42:066 Common/FileUtil.cpp:205 I[COMMON]: CreateFullPath: path /storage/savefiles/User/Load/Textures/
[... cut ...]
[libretro INFO] 24:42:066 Common/FileUtil.cpp:209 I[COMMON]: CreateFullPath: path exists /storage/savefiles/User/Dump/Frames/
[libretro INFO] 24:42:066 Common/FileUtil.cpp:205 I[COMMON]: CreateFullPath: path /storage/savefiles/User/Dump/Objects/
[libretro INFO] 24:42:066 Common/FileUtil.cpp:209 I[COMMON]: CreateFullPath: path exists /storage/savefiles/User/Dump/Objects/
[libretro INFO] 24:42:066 Common/FileUtil.cpp:205 I[COMMON]: CreateFullPath: path /storage/savefiles/User/Dump/Textures/
[libretro INFO] 24:42:066 Common/FileUtil.cpp:209 I[COMMON]: CreateFullPath: path exists /storage/savefiles/User/Dump/Textures/

dmrlawson, 2021-06-14:
Thanks.

```
[libretro ERROR] 24:41:968 VideoBackends/Vulkan/VulkanLoader.cpp:219 E[Video]: (CreateDevice) vkCreateDevice failed: (-7: VK_ERROR_EXTENSION_NOT_PRESENT)
```

It looks like the core is updated and now relies on something that is not (yet?) available on rpi4 :(

ToKe79, 2021-06-14:
for now stick to gl video driver for dolphin

dmrlawson, 2021-06-26:
Looks like it doesn't work and I was mistaken to think it used to.

I don't think there's much we can do here.

From memory the standalone version of dolphin didn't work properly with vulkan a number of months ago so I don't hold much hope that this will either.

Maybe once the driver has matured some more this will magically work

closed 2021-06-26 by dmrlawson