libretro/Lakka-LibreELEC #1394: Thumbnail Boxart not appearing (RPi 4b) (Stable 3.1 aarch64)

Rstation4, 2021-06-15:
Raspberry Pi 4 B  (2 Gb)
Lakka OS version: Lakka-RPi4.aarch64-3.1

In past versions as well, despite the thumbnail boxart being in the correct folder and having the correct playlist label name it does not appear. the image file is in a .png format. The 2 images are "Pocky & Rocky.png" and "Pocky & Rocky 2.png" both are in the correct file path.

File path: LAKKA_DISK//thumbnails/Nintendo - Super Nintendo Entertainment System/Named_Boxarts

Playlist entery below:
{
      "path": "/storage/roms/Snes/Pocky & Rocky.zip#Pocky & Rocky.smc",
      "label": "Pocky & Rocky",
      "core_path": "DETECT",
      "core_name": "DETECT",
      "crc32": "00000000|crc",
      "db_name": "Nintendo - Super Nintendo Entertainment System.lpl"
    },
    {
      "path": "/storage/roms/Snes/Pocky & Rocky 2.zip#Pocky & Rocky 2.smc",
      "label": "Pocky & Rocky 2",
      "core_path": "DETECT",
      "core_name": "DETECT",
      "crc32": "00000000|crc",
      "db_name": "Nintendo - Super Nintendo Entertainment System.lpl"
    },

dmrlawson, 2021-07-01:
I suspect this is a Retroarch bug. Are you able to test on any other platform?

Rstation4, 2021-07-09:
I did some research and according to a  post that some characters like &*/:<>?|` are replaced with an underscore. So I made a duplicate of that thumbnail that replaced the & with a underscore and that worked. They should mention this somewhere in documentation.

closed 2021-07-23 by Ntemis