libretro/Lakka-LibreELEC #1397: (Switch) System freezes when trying to connect PS3 controller

brenodantas10, 2021-06-19:
I'm using Lakka-Switch.aarch64-nightly-20210619-cb6ac9c on my Switch and tried to use my PS3 controller with it by different methods and all of then results in a frozen state that sometimes reboots the Switch.

Methods:
 - USB cable while docked -> Makes the system not recognize any inputs (even connected joy-cons) then freezes and reboot
 - Handheld OTG cable -> Makes the system not recognize inputs (even attached joycons) then freezes and reboot
 - Terminal with bluetoothctl (with previous given Switch's mac address to PS3 controller) -> Switch insta-freeze and doesn't reboot

I couldn't turn on logging because retroarch just resets when i try to save configuration with logging to file turned on, but i managed to get cached bluetooth information from /storage/.cache/bluetooth folder of my ps3 controller

[bluetooth.zip](https://github.com/libretro/Lakka-LibreELEC/files/6679898/bluetooth.zip)

brenodantas10, 2021-06-19:
I don't have any other console controller to test if this problem is PS3 controller only

Ntemis, 2021-06-30:
@GavinDarkglider

GavinDarkglider, 2021-06-30:
I have no idea the state of the ps3/ps4 controller driver in the 4.9 kernel that nvidia is using. as for the controller stuff in retroarch in general, I dont have autoconf stuff enabled, as there is no autoconf for linux joycons in retroarch, it only has android joycon support, and I am not sure how to get around that. I am not sure, but I do know xbox 1 controllers need you do disable ertm before they will work, but I have no idea about ps3 controllers. The only controllers I am supporting is the joycons, anything other than that you are on your own.

closed 2021-06-30 by Ntemis