libretro/Lakka-LibreELEC #1398: Feature request : Allow me to install a closed source / proprietary driver by copying it on my Lakka disk (no ethernet / ssh access needed)

leskenazi, 2021-06-20:
- Which version of Lakka are you using?
Lakka 3.1, stable and latest nightly, launched from a live SD card
- What system hardware are you using?
Macbook Pro 13 (Retina, 13 inch, Early 2015) with AirPort Extreme Wifi Card, and no ethernet port / ssh access
- What did you do?
Tried to list available Wifi SSID and to connect to my home Wifi
- What did you expect to happen?
List of available Wifi SSID, connection to my home Wifi
- What happened instead?
"no networks found" message displayed, cannot connect to my home Wifi
Seems like my wifi chip is unsupported by brcmfmac.
But proprietary drivers are available as stated here : https://help.ubuntu.com/community/WifiDocs/Driver/bcm43xx
My request : Allow me to install a closed source / proprietary driver by copying it on my Lakka disk (no ethernet / ssh access needed)

gouchi, 2021-06-20:
Thank you for the report.

Please try to provide some [log](https://www.lakka.tv/doc/Troubleshooting-Lakka/#automated-collection-of-relevant-logs) so we can check if we can enable support of the wifi.

leskenazi, 2021-06-20:
Hi

Here a log file created with Lakka 3.2 and the createlog script.

Congratulations on contributing to this nice project, and thanks for your help.

Best regards,

Laurent
[log-2021-06-20-13.36.34.zip](https://github.com/libretro/Lakka-LibreELEC/files/6682454/log-2021-06-20-13.36.34.zip)

ToKe79, 2021-06-20:
Hi @leskenazi.
thanks to @gouchi for pointing out my mistake in the kernel-firmware package. Can you please test [this image which should include the required firmware](https://nightly.builds.lakka.tv/members/vudiq/generic64-brcm_fw_fix/)? If it will fix your issue, I will push the fix. Thank you!

leskenazi, 2021-06-20:
This works great, I managed to connect to my Wifi and to download some thumbnails. Thanks a lot @ToKe79 and @gouchi !

commit 3da012bb (2021-06-20): kernel-firmware: revert incorrect test

the original test was meant to not remove the pcie brcm firmwares for
both - 64 and 32 bit x86 architecture (when porting Lakka to LE), but
actually the result of the test is always true and these firmwares are
also removed for 64 bit x86 architecture, which is the only case when
they are needed (these pcie firmwares are not needed on 32 bit x86).

Fixes #1398
closed 2021-06-20 by ToKe79