libretro/Lakka-LibreELEC #1400: [Lakka 3.2 - GPi Case - pi zero w] Wifi -> Connection failed!

Ivibaya, 2021-06-22:
**For GPi Case (Rpi Zero W)**

I am not sure if it's a bug or not, but I've tried with different SSID, no luck.

This is what I got in my error dump:
Apr 11 18:44:26 Lakka wpa_supplicant[448]: wlan0: Failed to initiate sched scan
Apr 11 18:44:27 Lakka wpa_supplicant[448]: wlan0: Trying to associate with SSID 'Home Network'
Apr 11 18:44:33 Lakka wpa_supplicant[448]: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=80:3f:5d:a3:e8:ea status_code=16
Apr 11 18:44:33 Lakka retroarch[477]: Error /net/connman/service/wifi_b827eb95794d_52656c6179204d4c494e43_managed_psk: Input/output error
Apr 11 18:44:37 Lakka wpa_supplicant[448]: wlan0: Trying to associate with SSID 'Home Network'
Apr 11 18:44:44 Lakka wpa_supplicant[448]: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=80:3f:5d:a3:e8:ea status_code=16

Does anybody experienced any similar WiFi issue?

PD: Screen rotation is perfect now :) Many thanks!

Ivibaya, 2021-06-23:
Tried to change SSID mode to b, g, n, mixed... No success.
Tried to set a simpler passphrase, success, but want to perform more network testings. Anyway, it was not necessary in Lakka 2.3, so maybe something is not good with the new drivers.

SethBarberee, 2021-07-31:
Have you tried restarting retroarch after connection failed? At least for me, this causes a IP address to populate and connect. In fact, I'm downloaing the newest Lakka update with it right now.

RobLoach, 2022-02-21:
HAVE_WIFI was just restored in the latest builds. Mind re-testing?

Ivibaya, 2022-02-21:
Yes, but sometimes it takes some retries until getting connected. Also had
to manage AP to make it work in a less noised channel, simpler password and
2,4GHz band.

closed 2022-06-10 by Ntemis