libretro/Lakka-LibreELEC #1401: Lakka 3.2 extend SD card partition error

atlus666, 2021-06-22:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using? =>  Lakka-RPi4.aarch64-nightly-20210621-3da012b
- What system hardware are you using?  => Raspberry Pi 4 4gb + 128 GB SD card
I burn img to SD Card , When the first boot , It resize partition and stop at reboot
It can't reboot
I force reboot it and enter the terminal to show mount
the /storage is still not extend ,but in parted ,the mmcblk0p2 already have 127GB

but this issue not happen in 64gb sd card  , maybe in bigger SD card extend script have problem?

[image]
It can;t reboot ,Just stop there, but no other message so I don't know what happend in bakground
[image]
After force reboot  
[image]
/storage only have 27mb
[image]
[image]
in parted , mmblk0p2 have 127gb

dmrlawson, 2021-07-01:
Do you have a spare 128GB sd card to test?

I have it running on a 128GB sd card but I can't say I've tested it from scratch on it (and I'd really rather not format it)

dmrlawson, 2021-07-06:
AFAIK this is not a common issue and I suspect the SD card is faulty

closed 2021-07-06 by dmrlawson