libretro/Lakka-LibreELEC #1402: Lakka 3.1 unable to upgrade to 3.2

MessiahMozgus, 2021-06-22:
I was perfectly able to run the online update from 3.0 to 3.1 a couple weeks ago. I barely used the Pi3 since then but decided to update again today. It downloads the 3.2 update to 100%, sits there for a while, then freezes (including the background animation) for a few minutes while still showing "downloaded 100%", as if installing, but then it just becomes responsive again with no indication of anything. The build info still shows 3.1 even after rebooting.

Rstation4, 2021-06-22:
@MessiahMozgus If it does not work then in the mean time you can do a more manual alternative of updating by downloading the Lakka 3.2 .img file and placing it in the the update folder of your Lakka OS Raspberry Pi 3 device. If you don't want to do it via samba or ssh you can use a Linux distro to copy and paste it in the update folder, you may need to use root permissions to add it to the folder.

ToKe79, 2021-07-13:
Try re-downloading the .tar file using the online updater for Lakka in RetroArch, After the download is finished, reboot your device - the update file should be applied **during** the reboot (not sure if this works with NOOBS).

closed 2021-07-23 by Ntemis