libretro/Lakka-LibreELEC #1407: RG351P/M Battery Level Issues on Lakka 3.2

taquitoboy, 2021-06-27:
- Which version of Lakka are you using?
Lakka 3.2

- What system hardware are you using?
RG351P and RG351M

- What did you do?
Deplete battery, recharge device while the device is powered off until LED indicator is turned off, and check battery levels.

- What did you expect to happen?
Battery level should be 100%.

- What happened instead?
Battery level maxes out at 12% - 13%. While depleting the battery again, it sits at 1% for the longest time.
Occurs on both of my RG351P and RG351M

ToKe79, 2022-04-01:
refers to discontinued branch, only updates to RetroArch / libretro-cores. As a workaround try using .dtb from 351ELEC.

closed 2022-04-01 by ToKe79

Unix0815, 2022-05-11:
> refers to discontinued branch, only updates to RetroArch / libretro-cores. As a workaround try using .dtb from 351ELEC.

I'm using the latest Lakka 3.7 with the 351elec Dr on my rg351p. At first, the battery was showing correct values. But after charging over night, it maxes out at 31%, even though the battery should be fully charged.

Any fix for this behavior? Where can I download the 4.0 release for the rg351p?

Unix0815, 2023-04-27:
Will this ever be fixed?

Ntemis, 2023-04-27:
not on 3.2 for sure