libretro/Lakka-LibreELEC #1408: [Feature request] Support for HDMI-CEC

tomillr, 2021-06-27:
It's about time that Lakka adds support for the HDMI-CEC standard. CEC provides a lot of quality of life improvements and makes a machine running Lakka act more like a of the shelf media device.

For example, with CEC you can automatically have your TV and AVR be turned on and switched to the proper HDMI input just by turning on the Lakka machine. Also, you can use the TV remote to navigate the menus

There’s support for HDMI-CEC in the Linux kernel starting with v4.8: 
https://01.org/linuxgraphics/gfx-docs/drm/media/kapi/cec-core.html

The HDMI-CEC adapters made by Pulse Eight (internal boards as well as a USB dongle) can be used to add support for CEC on x86/x64 machines: 
https://www.pulse-eight.com/p/104/usb-hdmi-cec-adapter

Many tinker boards like the Raspberry Pi or the last few generations of Intel NUCs already come with an CEC adapter as part of the SoC. It's also been a feature of popular HTPC software like Kodi for many years.

Ntemis, 2021-06-30:
Lakka doesnt need CEC support for various reasons:
1. It doesn't use CEC anywhere its a gaming console after all
2. Enabling CEC delays the boot up as its looking for a CEC compatible adapter.

closed 2021-06-30 by Ntemis

tomillr, 2021-06-30:
> Lakka doesnt need CEC support for various reasons:
> 
> 1. It doesn't use CEC anywhere its a gaming console after all
> 2. Enabling CEC delays the boot up as its looking for a CEC compatible adapter.

Not sure I understand your reasoning.

1. If a console doesn't need a feature like HDMI-CEC, then why do a Playstation 3, 4, 5,  a Xbox Series S/X or even a Nintendo Switch support this? 

This really is a quality of life improvement for everyone with compatible equipment (which is basically every modern tv sold in the last 8 years). Why can't we have Lakka automatically turn on the TV (plus a connected AVR) and have the device automatically switch the HDMI channels on the TV and AVR?

2. Just compare how it's implemented in Kodi for example: It's an optional feature disabled by default. If your device doesn't support HDMI-CEC you'd properly never even enable it and thus don't have to deal with any delays.

dmrlawson, 2021-06-30:
> This really is a quality of life improvement for everyone with compatible equipment (which is basically every modern tv sold in the last 8 years). Why can't we have Lakka automatically turn on the [...]

You can enable this behaviour by changing [this line in config.txt](https://github.com/libretro/Lakka-LibreELEC/blob/master/projects/RPi/devices/RPi4/config/config.txt#L28), I think it's disabled for the reason @Ntemis mentioned.

The original issue sounded like a desire to control the menu with the tv remote which is something that's probably not going to be worked on.

tomillr, 2021-06-30:
> You can enable this behaviour by changing [this line in config.txt](https://github.com/libretro/Lakka-LibreELEC/blob/master/projects/RPi/devices/RPi4/config/config.txt#L28), I think it's disabled for the reason @Ntemis mentioned.
Interesting, never have come across that line so far. Wouldn't it make sense to surface this setting as a simple on/off switch in the settings menu?