libretro/Lakka-LibreELEC #1409: performance regression. latter nightlies lower peformence on cores with higher internal resolutions

auxfam, 2021-06-28:
This issue tracker is NOT for help or support. Please use one of the following channels for help and support:
- the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general)
- via IRC @ #lakkatv on freenode.net
- #lakkatv on libretro discord https://discord.gg/retroarch

Before you report a bug make sure you have tried the latest nightly version of Lakka at https://nightly.builds.lakka.tv/latest/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
Lakka-Generic.x86_64-nightly-20210625-a4ff855.img.gz

- What system hardware are you using?
intel NUC8 Intel® Core™ i5-8259U Processor (6M Cache, up to 3.80 GHz)
note: this has 128 EDRAM as a L4 cpu/GPU cache

- What did you do?
installed latest nighties as above
run flyca\st
with gl backend (also tested with glcore)
load Dead or Alive
let intro run
play a couple of cpu matches

- What did you expect to happen?
in game match (v cpu) is expected to perform around 58-60.  when with following deviations from default
Internal resolution: 1280x960
accuracy: high

- What happened instead?
intro sees drops to 40+ fps
matches sees drops to 40+ fps

- other notes:

cpu gov: max performance (also tried per core max performence)
monitored /proc/cpuinfo whilist testing and cpu clockspeed stays at max(3800 MHz or so) so gov looks like it's working
reduce resolution back down to 640x480 brings fps back up to a fairly consistant 60
my previous settings on latency for 8/16 bit cores mostly had a delay frame of 8-9mseconds after installing the latest nightly these started showing some more nuance drops (solid 60.02 to 58 spikes) so had to dial that back to 3-4ms
I had to roll back to this build
`Lakka-Generic.x86_64-nightly-20210615-2a94fae.img.gz`
to restore prior perfomence (slightly later build might be ok but this is the one which felt right)

given that cpu is not throttle afaik.  and it is most noticable on shader, higher internal resolutions this feels like a GPU perfomence  regression of some sort.

gouchi, 2021-06-30:
Will it possible to get some [log](https://www.lakka.tv/doc/Troubleshooting-Lakka/#automated-collection-of-relevant-logs) with `createlog` ?

Thank you.

auxfam, 2021-07-01:
I;ve updated to https://nightly.builds.lakka.tv/2021-07-01/Generic.x86_64/Lakka-Generic.x86_64-nightly-20210701-7c59da8.img.gz
for the following logs

`[INFO] RetroArch 1.9.5 (Git 6738fa57aa)
[INFO] === Build =======================================
[INFO] CPU Model Name: Intel(R) Core(TM) i5-8259U CPU @ 2.30GHz
[INFO] Capabilities:  MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES AVX
[INFO] Built: Jun 19 2021
[INFO] Version: 1.9.5
[INFO] Git: 6738fa57aa
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [DRM]: Found 3 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 20 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Connector 2 connected: no
[INFO] [DRM]: Connector 2 has 0 modes.
[INFO] [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 1: (1920x1080) 1920 x 1080, 59.940201 Hz
[INFO] [DRM]: Mode 2: (1920x1080i) 1920 x 1080, 30.000000 Hz
[INFO] [DRM]: Mode 3: (1920x1080i) 1920 x 1080, 29.970100 Hz
[INFO] [DRM]: Mode 4: (1920x1080i) 1920 x 1080, 25.000000 Hz
[INFO] [DRM]: Mode 5: (1280x1024) 1280 x 1024, 60.019741 Hz
[INFO] [DRM]: Mode 6: (1440x900) 1440 x 900, 59.901459 Hz
[INFO] [DRM]: Mode 7: (1360x768) 1360 x 768, 60.015163 Hz
[INFO] [DRM]: Mode 8: (1280x720) 1280 x 720, 60.000000 Hz
[INFO] [DRM]: Mode 9: (1280x720) 1280 x 720, 59.940201 Hz
[INFO] [DRM]: Mode 10: (1280x720) 1280 x 720, 50.000000 Hz
[INFO] [DRM]: Mode 11: (1024x768) 1024 x 768, 60.003841 Hz
[INFO] [DRM]: Mode 12: (1440x480i) 1440 x 480, 29.970030 Hz
[INFO] [DRM]: Mode 13: (800x600) 800 x 600, 60.316540 Hz
[INFO] [DRM]: Mode 14: (720x576) 720 x 576, 50.000000 Hz
[INFO] [DRM]: Mode 15: (720x480) 720 x 480, 60.000000 Hz
[INFO] [DRM]: Mode 16: (720x480) 720 x 480, 59.940060 Hz
[INFO] [DRM]: Mode 17: (640x480) 640 x 480, 60.000000 Hz
[INFO] [DRM]: Mode 18: (640x480) 640 x 480, 59.940475 Hz
[INFO] [DRM]: Mode 19: (464x480) 464 x 480, 85.429520 Hz
[INFO] [GL]: Found GL context: kms
[INFO] [GL]: Detecting screen resolution 1920x1080.
[INFO] [EGL] Found EGL_EXT_platform_base, trying eglGetPlatformDisplayEXT
[INFO] [EGL]: EGL version: 1.5
[INFO] [EGL]: Current context: 0x2a84230.
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [GL]: Vendor: Intel, Renderer: Mesa Intel(R) Iris(R) Plus Graphics 655 (CFL GT3).
[INFO] [GL]: Version: 4.6 (Compatibility Profile) Mesa 21.1.4.
[INFO] [GL]: Using resolution 1920x1080
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[INFO] [GLSL]: Checking GLSL shader support ...
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GL]: Using 4 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [GL]: Using GL_RGB565 for texture uploads.
[... 30 lines cut ...]
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [CORE]: Using content: /storage/roms/int-games/Sega_-_Dreamcast/Dead_or_Alive_2_-jp-__-Shokai_Genteiban-_.chd.
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
[INFO] [Video]: Does not have enough samples for monitor refresh rate estimation. Requires to run for at least 4096 frames.
[INFO] RetroArch 1.9.5 (Git 6738fa57aa)
[INFO] === Build =======================================
[INFO] CPU Model Name: Intel(R) Core(TM) i5-8259U CPU @ 2.30GHz
[INFO] Capabilities:  MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES AVX
[INFO] Built: Jun 19 2021
[INFO] Version: 1.9.5
[INFO] Git: 6738fa57aa
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Core]: Loading dynamic libretro core from: "/tmp/cores/flycast_libretro.so"
[INFO] [Overrides]: Core-specific overrides found at "/storage/.config/retroarch/config/Flycast/Flycast.cfg".
[INFO] [Overrides]: No content-dir-specific overrides found at "/storage/.config/retroarch/config/Flycast/Sega_-_Dreamcast.cfg".
[INFO] [Overrides]: No game-specific overrides found at "/storage/.config/retroarch/config/Flycast/Dead_or_Alive_2_-jp-__-Shokai_Genteiban-_.cfg".
[INFO] [Config]: Appending config "/storage/.config/retroarch/config/Flycast/Flycast.cfg".
ALSA lib /home/lakkabuildbot/nightly/Lakka-LibreELEC/build.Lakka-Generic.x86_64/alsa-lib-1.2.3.2/src/seq/seq_hw.c:466:(snd_seq_hw_open) open /dev/snd/seq failed: No such file or directory
ALSA lib /home/lakkabuildbot/nightly/Lakka-LibreELEC/build.Lakka-Generic.x86_64/alsa-lib-1.2.3.2/src/seq/seq_hw.c:466:(snd_seq_hw_open) open /dev/snd/seq failed: No such file or directory
`

the last few lines are after starting up the game

auxfam, 2021-07-01:
additionaly the video and audio logs

> 00:02.0 VGA compatible controller [0300]: Intel Corporation Iris Plus Graphics 655 [8086:3ea5] (rev 01)
        DeviceName:  CPU
        Subsystem: Intel Corporation Iris Plus Graphics 655 [8086:2074]
        Kernel driver in use: i915

> null
    Discard all samples (playback) or generate zero samples (capture)
default:CARD=PCH
    HDA Intel PCH, ALC233 Analog
    Default Audio Device
sysdefault:CARD=PCH
    HDA Intel PCH, ALC233 Analog
    Default Audio Device
front:CARD=PCH,DEV=0
    HDA Intel PCH, ALC233 Analog
    Front output / input
surround21:CARD=PCH,DEV=0
    HDA Intel PCH, ALC233 Analog
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=PCH,DEV=0
    HDA Intel PCH, ALC233 Analog
    4.0 Surround output to Front and Rear speakers
surround41:CARD=PCH,DEV=0
    HDA Intel PCH, ALC233 Analog
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=PCH,DEV=0
    HDA Intel PCH, ALC233 Analog
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=PCH,DEV=0
    HDA Intel PCH, ALC233 Analog
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=PCH,DEV=0
    HDA Intel PCH, ALC233 Analog
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
hdmi:CARD=PCH,DEV=0
    HDA Intel PCH, HDMI 0
    HDMI Audio Output
hdmi:CARD=PCH,DEV=1
    HDA Intel PCH, HDMI 1
    HDMI Audio Output
hdmi:CARD=PCH,DEV=2
    HDA Intel PCH, HDMI 2
    HDMI Audio Output
hdmi:CARD=PCH,DEV=3
    HDA Intel PCH, HDMI 3
    HDMI Audio Output
hdmi:CARD=PCH,DEV=4
    HDA Intel PCH, HDMI 4
    HDMI Audio Output

auxfam, 2021-07-01:
[debu-createlog.zip](https://github.com/libretro/Lakka-LibreELEC/files/6748004/debu-createlog.zip)

a more detailed set using `createlog`
this is me booting up and runnig the game for a round

Ntemis, 2021-07-01:
found your issue
https://github.com/libretro/Lakka-LibreELEC/pull/1392

auxfam, 2021-07-01:
is this in refernce to the `Alpha Sorting` option?   this has always been set to `Per-Pixel (accurate, but slowest)`  .  did the this setting previous actually did nothing prior to the above commit?   flipping it down to to `Per-Triangle (normal)` does indeed improve the framerate (but still seems to fall short a couple of frames overall).   plus other cores seem to be slower too?   notably ppsspp Tekken6 (bufferchain:2, 1020x1088 render target, 4x aniso) was 58-60 (with drops) is now 50 -57.

gouchi, 2021-07-07:
@Ntemis nice catch ! I forgot we enabled it.

@auxfam Yes I think so, I remember correctly. If needed the [compatibility list for OIT](https://github.com/libretro/flycast/issues/136).

For ppsspp maybe it is only for some [games](https://github.com/hrydgard/ppsspp/issues/13260) but it is for RPi4 not x86_64 ?

auxfam, 2021-07-07:
it's unclear overall.   I can only report the titles I replay frequently and they all have small regressions that you may not normally notice were it not for the fact that I tuned most of my core configs to sweet spots,  unfortunately we have no reports from other users to corroborate this or otherwise.   in general this just feels like an increase in performence requirements which for a project of this size can be expected over time.

I am ok with that being the case (in which case close this issue)  but last thing I can say is that it's not just the x86 platform.   I fired up my RG350p (ARM64) last week on the PSCx-ReArmed core and tried Tekken 3 (PS1) [ max 3 swap buffer, power profile:max, threaded video: defaults on everything else ] ~ 58-60 ,  [Thread video:off] 53
for ReArmed this feel far too low / unstable but again haven't had the time to go back and test older nightlies yet.

gouchi, 2021-07-11:
If it is on other platform it could the update of the core or graphic stack.

Please reopen this issue or open a new one if the performance is getting worse.

Thank you for your tests and report.

closed 2021-07-11 by gouchi