libretro/Lakka-LibreELEC #1412: RG351P/M possile improvement(battery display,  led light,  sleep mode,  OTG port)

Zalica, 2021-06-30:
- Which version of Lakka are you using?
 Lakka 3.2

- What system hardware are you using?
 RG351M

I just got  RG351M a week ago. Generally Lakka worked quite fine with it.But there are something I noticed:
1. Battry problem.
 the battry level problem,same as https://github.com/libretro/Lakka-LibreELEC/issues/1407

2.LED problem
there is led light near the X button,according to the menu,grean means power on, red means battry charging.However,Lakka 3.1 and 3.2 made the led always showing both green and red,thus the led just can't show any useful infomation.

3. Sleep mode would be useful.
Short click on the power button does nothing.On the other hand,in the system shipped with it ,short click on the power button,the machin goes into sleep mode. I think a sleep mode somehow follows the instinct more properly.

4.OTG port for external usb storage is quit good.
The OTG port is quite convenient,but when I connected the usb stick,lakka can't find it. I can't find any new device or partrition when loading content.I think it will be very good If I can use a extra usb stick.

I'm just a plain user,I've been using retroarch/lakka for quite sometime, I'm not rushing anything ,just hope lakka gets better.

Thank you for the great work.

Shadow53, 2021-11-02:
I have the RG351MP and would also like to see this, especially the sleep mode. I'm willing to try my hand at it, but I am not sure where to start

Ntemis, 2021-11-02:
@davidgfnet any updates on this issue?

zongyuanliu-bio, 2021-11-07:
> I have the RG351MP and would also like to see this, especially the sleep mode. I'm willing to try my hand at it, but I am not sure where to start

I can not even boot, did you guys have similar issue?

davidgfnet, 2021-11-12:
Currently focusing the efforts on getting this up and running on the new Lakka v4 branch, perhaps we can backport stuff later. There's better kernel support for many of the issues explained here.

GavinDarkglider, 2021-11-15:
> Currently focusing the efforts on getting this up and running on the new Lakka v4 branch, perhaps we can backport stuff later. There's better kernel support for many of the issues explained here.

For the sleep issue, you might want to check what keyboard button the kernel redirects that command to. I had to patch dts in L4T builds to correct this, or else everytime you pushed the button it would shutdown.

ToKe79, 2022-04-01:
as the focus is currently to bring RK3266 support to the Lakka 4.x branch (with mainline kernel), we will not be fixing old bugs. thank you for your understanding.

closed 2022-04-01 by ToKe79