libretro/Lakka-LibreELEC #1415: NOOBS documentation update

ItachiSan, 2021-07-02:
The page:
https://www.lakka.tv/doc/Alternative-installation-methods/
provides out-of-date information on how to install Lakka with NOOBS.

I've updated the documentation in a separate copy:
https://github.com/ItachiSan/Lakka-Wiki/commit/5e545fbdee40781bd55c629eb07b28aa22d6a429
As I see no option for creating a pull request for just the wiki, I decided to open an issue for discussing how to deal with it. 😄

gouchi, 2021-07-13:
Thank you it is has been [updated](https://github.com/libretro/Lakka-LibreELEC/wiki/Alternative-installation-methods).

@ToKe79 When update the website please update submodules docs ;)

closed 2021-07-26 by Ntemis

ItachiSan, 2021-07-26:
This should not be closed as the website submodule was not updated.

gouchi, 2021-08-01:
[Website](https://www.lakka.tv/doc/Alternative-installation-methods/) is up to date now.