libretro/Lakka-LibreELEC #1416: Create possibility to remove bluetooth device in the Retroarch frontend

coldspark29, 2021-07-04:
- Which version of Lakka are you using?

857b254

- What system hardware are you using?

Pi4

- What did you do?

I paired my PS5 controller with another device after using it with Lakka. I had to manually call ``bluetoothctl`` and remove it. Then repair.

ToKe79, 2021-09-16:
These images contain a patch to RetroArch that adds the possibility to remove pairing by pressing `Start` button (or `Space` on keyboard) on the listed device, which should be removed.
https://nightly.builds.lakka.tv/members/vudiq/bt_remove_device/

ToKe79, 2021-09-16:
Note: if the device, for which the pairing was removed, is currently not connected, there is no visual feedback, the removal happens in the "backend" :-)

ToKe79, 2021-09-17:
fixed by https://github.com/libretro/Lakka-LibreELEC/commit/86441fdd056a01bbe3483a21e772f372a2813440

closed 2021-09-17 by ToKe79