libretro/Lakka-LibreELEC #1418: Raspberry Pi2 + Netgear Wifi Dongle : No SSID detected (works with Raspbian / Raspberry Pi OS)

leskenazi, 2021-07-06:
- Which version of Lakka are you using?
Lakka-RPi2.arm-nightly-20210705-857b254
- What system hardware are you using?
Raspberry Pi2 + Netgear Wifi Dongle
- What did you do?
Tried to list wifi networks available
- What did you expect to happen?
List of SSIDs
- What happened instead?
No network detected

Wifi works perfectly with Raspbian / Raspberry Pi OS.

Please see attached log file.
[log-2019-04-11-16.28.55.zip](https://github.com/libretro/Lakka-LibreELEC/files/6771077/log-2019-04-11-16.28.55.zip)

Ntemis, 2021-07-06:
[    4.694902] usb 1-1.4: Product: 802.11n WLAN Adapter
[    4.694921] usb 1-1.4: Manufacturer: Realtek
We have no power over quality of usb chips. Your wifi is recognized just fine but if it cant find any SSID's its range is an issue we cant fix

leskenazi, 2021-07-06:
Are you sure ? That's strange because I also runs Raspberry Pi OS with the same hardware (and at the same place) and it detects all SSIDs perfectly.

Ntemis, 2021-07-06:
The wifi is working fine from the logs you sent.

leskenazi, 2021-07-06:
These are the networks I can connect to from Raspberry Pi OS :

[image]
[image]

And this is the network I can connect to from Lakka (same hardware / location) :

[image]
[image]

I checked Lakka configuration, everything seems fine to me.

What may I be missing ?

Ntemis, 2021-07-07:
Give me a dmesg log

leskenazi, 2021-07-07:
Thanks for your help. Here is a dmesg log.

Best regards,

Laurent
[dmesglog.txt](https://github.com/libretro/Lakka-LibreELEC/files/6776843/dmesglog.txt)

gouchi, 2021-07-07:
Don't know if they have used the [kernel driver](https://cateee.net/lkddb/web-lkddb/RTL8192CU.html) or the [outside kernel driver](https://github.com/libretro/Lakka-LibreELEC/tree/master/packages/linux-drivers/RTL8192CU) ?

leskenazi, 2021-07-08:
Hi @gouchi. I suppose your message is for @Ntemis, right ?

gouchi, 2021-07-08:
Hi @leskenazi, everybody so that we can advance to fix this issue ;)

closed 2021-07-26 by Ntemis