libretro/Lakka-LibreELEC #1419: [Feature Request] Wireguard VPN

MaverickTN, 2021-07-07:
- Which version of Lakka are you using?
- v3.2
- What system hardware are you using?
- ODroid XU4, ODroid H2/H2+
- What did you do?
- N/A
- What did you expect to happen?
- N/A
- What happened instead?
- N/A

Would be nice to have Wireguard VPN so that remote file shares can be accessible when remote. I've found a way to do it for now, but I am using a small travel router that has Wireguard support in order to manage it.  I noticed that the nightly images have wireguard as a folder in /storage/.config/wireguard. Was this an intended feature or something that got randomly thrown in?

ToKe79, 2021-08-05:
Unfortunately, the kernel version (4.14) we use for OdroidXU3/4 does not have the required config flag (`CONFIG_WIREGUARD`) to enable WireGuard.

MaverickTN, 2021-08-05:
Wireguard has a kernel module that is available for kernels prior to 5.8.

ToKe79, 2021-09-13:
@MaverickTN Hi, please test dev image from http://nightly.builds.lakka.tv/PRs/pr_1499_000429/. Thank you

closed 2021-09-17 by Ntemis

MaverickTN, 2021-09-17:
> @MaverickTN Hi, please test dev image from http://nightly.builds.lakka.tv/PRs/pr_1499_000429/. Thank you

My cousin has this device for the moment, I will try it in a few days when I get it back. Is this also included in the x86 image? I can try it on my Odroid H2.

ToKe79, 2021-09-18:
It it included in all images, except Switch.

ToKe79, 2021-09-18:
Also the pull request was merged, so you may also test with [latest nightly build](https://nightly.builds.lakka.tv/latest/).

PR #1499 (2021-09-17): add wireguard support to Lakka
fixes #1419