libretro/Lakka-LibreELEC #1420: No Audio through HDMI port.

Kei84, 2021-07-08:
LAKKA 3.2
I have tried multiple combinations of the "Audio" & "Device".
I have an HDMI cable hooked up
Here is the link to my log
https://transfer.sh/1zZb17r/log-2021-07-08-02.56.57.zip

Kei84, 2021-07-08:
So after, updating Bios. using 3.5mm jack, changing out HDMI cables, editing config files. I finally found out what is happening.
When the computer is turned off and hdmi is plugged into computer, if you turn on the computer load up in Lakka. you get video but no sound. to fix the no sound issue. while the computer is still powered on into lakka. unplug the hdmi cable from the pc and plug it back in. sound works. this is a band-aid fix, why would lakka not initialize the HDMI sound without me having to unplug and plug the hdmi cord back in while running lakka?

Ntemis, 2021-07-08:
Is not lakka's fault. Report the issues to Raspberry Pie foundations github.
Its something they slipped by about your specific hardware(tv,monitor) so report tell them and pray to be able to fix it

Kei84, 2021-07-08:
> Is not lakka's fault. Report the issues to Raspberry Pie foundations github.
> Its something they slipped by about your specific hardware(tv,monitor) so report tell them and pray to be able to fix it

I'm not using a raspberry pi. There is an initialization failure when first booting up with the audio device. once fully botted up and able to move around the menu, i then unplug the hdmi cable and re plug it back in. then making lakka initialize the audio for the hdmi. I do not see where Raspberry Pie Foundation would be helpful in this as i am usuing a generic PC board. "ga-ma785gm us2h"

Ntemis, 2021-07-08:
It's a handshake issue between your tv and your gpu.
What is the GPU model?

gouchi, 2021-07-08:
The integrated GPU is `[AMD/ATI] RS880 [Radeon HD 4200] [1002:9710]` which is [R600 gpu family](https://www.x.org/wiki/RadeonFeature/).

You should have for the audio device `hdmi:CARD=HDMI,DEV=0`.

Several things to try

- Try to power on the TV before starting your PC.
- Try to append to [cmdline](https://www.lakka.tv/doc/The-bootloader/) `radeon.audio=1`

Thank you.

Kei84, 2021-07-08:
> The integrated GPU is `[AMD/ATI] RS880 [Radeon HD 4200] [1002:9710]` which is [R600 gpu family](https://www.x.org/wiki/RadeonFeature/).
>
> You should have for the audio device `hdmi:CARD=HDMI,DEV=0`. [...]

I've tried with the tv on first, and then turning it on after lakka boots into it.
Haven't tried the cmdline radoen.audio=1

I have also noticed that if i have a my yellow USB port(the only one at the time) being used, it won't work with my band-aid fix, i had to plug it into one of my lower black usb ports for my band-aid fix to work.

gouchi, 2021-07-08:
> I just tried the radeon.audio=1.
> it returned "-sh: radeon.audio=1: not found"

You did append the line as [specified in the documentation](https://www.lakka.tv/doc/The-bootloader/#on-generic-pc) ?

It looks like you launched it in the shell which you should not.

Thank you.

Kei84, 2021-07-09:
> > I just tried the radeon.audio=1.
> > it returned "-sh: radeon.audio=1: not found"
> 
> You did append the line as [specified in the documentation](https://www.lakka.tv/doc/The-bootloader/#on-generic-pc) ?
> 
> It looks like you launched it in the shell which you should not.
> 
> Thank you.

These are my two files.
/flash/syslinux.cfg
================
DEFAULT linux
PROMPT 0

LABEL linux
 KERNEL /KERNEL
 APPEND boot=UUID=327D-1AD0 disk=UUID=02cf8070-3dfb-4c82-abde-a109f781f000  quiet
================
/flash/EFI/BOOT/grub.cfg
================
set timeout="0"
set default="LibreELEC"

menuentry "LibreELEC" {
        search --set -f /KERNEL
        linux /KERNEL boot=UUID=327D-1AD0 disk=UUID=02cf8070-3dfb-4c82-abde-a109f781f000 quiet
}
================

do i just add it after the "quiet"?

I'm sorry i am not familiar with editing these files with these commands.

gouchi, 2021-07-09:
Yes please add the parameter `radeon.audio=1` at the end of the line after quiet.

Kei84, 2021-07-09:
> Yes please add the parameter `radeon.audio=1` at the end of the line after quiet.

That did not work. as i still have to turn tv off and then back on after lakka booted up.

Found out just turning the tv off and back on again after it boots up made the audio work instead of just having to unplug the hdmi cable.

gouchi, 2021-07-10:
Hi,

Thank you for your feedback. I am glad you find a workaround.

I am not sure if we can fix it.  I saw this error in the log

`snd_hda_codec_hdmi hdaudioC1D0: HDMI ATI/AMD: no speaker allocation for ELD`

It seems the issue is maybe related to the fact that radeon driver does not implement [audio component](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1834771)(comment 54). There were [two](https://lists.freedesktop.org/archives/dri-devel/2018-July/184240.html) [attempts](https://lists.freedesktop.org/archives/dri-devel/2019-July/227423.html) but so far patches have  not been merged.

Will it possible to get some log from before and after your turn on/off TV ?

`cat /proc/asound/card1/eld#0.*`
`cat /sys/module/snd_hda_codec_hdmi/parameters/static_hdmi_pcm`

And if you can check `alsamixer` also for the hdmi (Press F6 to select HDA ATI HDMI)

Thank you.

Ntemis, 2021-07-27:
Its a hardware specific bug that no upstream fix exist on linux side. So there is nothing to be fixed here actually.
Report the issue on linux mailing list for a possible solution on kernel side.
As for lakka issues is now considered closed.

closed 2021-07-27 by Ntemis