libretro/Lakka-LibreELEC #1422: [Feature Request] Native image viewer

elShalafi, 2021-07-08:
Hello
I'm using Lakka 3.2 on RaspberryPi4B
I wish to ask for a enhancement.
I was playing with the Images feature (the playlist Images), figuring out how it works. As far as I can see, in order to view/show an image Lakka is using a "pseudo-core". This means that before opening an image any other running core/content has to be closed (because only one content can be loaded at the same time)
Use case: I am playing a RPG game and I have the maps of the game also in Lakka. I would like to be able to pause the game (without closing the content... just "quick menu"), navigate to the Images, consult the map, and return to the game.
Until now, as I said, the game is ended as soon as the map is opened. This is what I would like to avoid.

I know that I can save a state, close the game, consult the map, re-run the game and load the state... But I would like something more... seamless

I just suggest to use a native image viewer, directly managed by the OS. Consequently, avoiding using a "pseudo-core" and leaving libreto/retroarch in stand by

Thank you so much!

gouchi, 2021-07-09:
Thank you the feature request but it will be better if you opened it in [RetroArch issue tracker](https://github.com/libretro/RetroArch/issues).

I am not sure if it will be possible to have the preview image in fullscreen.

[image]

elShalafi, 2021-07-09:
Thank you for your answer. I'll open it in RetroArch issue tracker too.

About the preview workaround... yes, actually the image is previewed as a thumbnail as usual, but if you think about my use case, this is not practical. I mean, the map of a RPG game is usually very large and you will see nothing in a preview. Probably you need a magnifier feature indeed...

djleon1983, 2021-07-09:
indeed.
i also wish there will be a way to view the pdf/txt game manual in lakka/retroarch.

elShalafi, 2021-07-09:
Hello again
Woow! A friend of mine just told me something I didn't know about the Lakka UI: if you push START when navigating along the items of a playlist, the thumbnail pops up and it is shown almost at full screen (no need to load any core). This addressed almost 100% of my request. The only remaining feature would be the magnifier. Being able to zoom in over the image and, obviosly, being able to move over it as long as the image becomes larger than the screen.

gouchi, 2022-03-02:
Closing this issue as it has been opened in RA issue tracker. Moreover, there is a workaround.

closed 2022-03-02 by gouchi