libretro/Lakka-LibreELEC #1430: retroarch stop to work when start run N64 on RPi3b+

frani, 2021-07-20:
This issue tracker is NOT for help or support. Please use one of the following channels for help and support:
- the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general)
- via IRC @ #lakkatv on freenode.net
- #lakkatv on libretro discord https://discord.gg/retroarch

Before you report a bug make sure you have tried the latest nightly version of Lakka at https://nightly.builds.lakka.tv/latest/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
RPi2.arm-3.2
- What system hardware are you using?
RPi3b+ 4gb ram
- What did you do?
try to run N64 rom
- What did you expect to happen?
play the rom
- What happened instead?
crash when star to run emulator

### 01_RetroArch.log
```
################################################################################
# ... output of retroarch -v
# Lakka release: RPi2.arm-3.2
#
################################################################################
wifi is already enabled
ALSA lib /home/lakkabuildbot/release/Lakka-LibreELEC/build.Lakka-RPi2.arm/alsa-lib-1.2.3.2/src/seq/seq_hw.c:466:(snd_seq_hw_open) open /dev/snd/seq failed: No such file or directory
wifi is already enabled
ALSA lib /home/lakkabuildbot/release/Lakka-LibreELEC/build.Lakka-RPi2.arm/alsa-lib-1.2.3.2/src/seq/seq_hw.c:466:(snd_seq_hw_open) open /dev/snd/seq failed: No such file or directory
unknown NIR ALU inst: vec1 32 ssa_124 = umin ssa_55, ssa_57
Aborted
```

### 02_System.log
```
################################################################################
# ... output of dmesg
# Lakka release: RPi2.arm-3.2
#
################################################################################
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.39 (lakkabuildbot@globox) (armv7ve-libreelec-linux-gnueabi-gcc-9.3.0 (GCC) 9.3.0, GNU ld (GNU Binutils) 2.33.1) #1 SMP Fri Jun 18 17:12:46 CEST 2021
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x1ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000] On node 0 totalpages: 196608
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s51788 r8192 d21940 u81920
[    0.000000] pcpu-alloc: s51788 r8192 d21940 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 194880
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  boot=UUID=1806-4614 disk=UUID=6a56e1de-2d03-4ad6-979d-05073f8a74ce quiet root=/dev/ram0 rdinit=/init usbcore.autosuspend=-1
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 496328K/786432K available (11264K kernel code, 1313K rwdata, 2488K rodata, 3072K init, 572K bss, 27960K reserved, 262144K cma-reserved)
[... 1037 lines cut ...]
Apr 11 16:28:48 Lakka bluetoothd[494]: Failed to set privacy: Rejected (0x0b)
Apr 11 16:28:48 Lakka retroarch[484]: wifi is already enabled
Apr 11 16:28:50 Lakka retroarch[484]: ALSA lib /home/lakkabuildbot/release/Lakka-LibreELEC/build.Lakka-RPi2.arm/alsa-lib-1.2.3.2/src/seq/seq_hw.c:466:(snd_seq_hw_open) open /dev/snd/seq failed: No such file or directory
Apr 11 16:28:50 Lakka avahi-daemon[433]: Joining mDNS multicast group on interface wlan0.IPv4 with address 192.168.0.18.
Apr 11 16:28:50 Lakka avahi-daemon[433]: New relevant interface wlan0.IPv4 for mDNS.
Apr 11 16:28:50 Lakka avahi-daemon[433]: Registering new address record for 192.168.0.18 on wlan0.IPv4.
Apr 11 16:28:50 Lakka connmand[409]: Setting domainname to supercanal.com.ar
Apr 11 16:28:50 Lakka connmand[409]: wlan0 {add} address 192.168.0.18/24 label wlan0 family 2
Apr 11 16:28:50 Lakka connmand[409]: wlan0 {add} route 192.168.0.0 gw 0.0.0.0 scope 253 <LINK>
Apr 11 16:28:50 Lakka connmand[409]: wlan0 {add} route 192.168.0.1 gw 0.0.0.0 scope 253 <LINK>
Apr 11 16:28:50 Lakka connmand[409]: wlan0 {add} route 8.8.8.8 gw 192.168.0.1 scope 0 <UNIVERSE>
Apr 11 16:28:50 Lakka connmand[409]: wlan0 {add} route 8.8.4.4 gw 192.168.0.1 scope 0 <UNIVERSE>
Apr 11 16:28:50 Lakka connmand[409]: wlan0 {add} route 0.0.0.0 gw 192.168.0.1 scope 0 <UNIVERSE>
Apr 11 16:28:54 Lakka nmbd[474]: [2019/04/11 16:28:54.451287,  0] ../lib/util/become_daemon.c:135(daemon_ready)
Apr 11 16:28:54 Lakka nmbd[474]:   daemon_ready: daemon 'nmbd' finished starting up and ready to serve connections
Apr 11 16:28:55 Lakka connmand[409]: ntp: adjust (jump): +71757502.471557 sec
Jul 20 05:07:23 Lakka connmand[409]: wlan0 {add} route 82.165.8.211 gw 192.168.0.1 scope 0 <UNIVERSE>
Jul 20 05:07:23 Lakka crond[294]: time disparity of 1195958 minutes detected
Jul 20 05:07:26 Lakka connmand[409]: wlan0 {del} route 82.165.8.211 gw 192.168.0.1 scope 0 <UNIVERSE>
Jul 20 05:07:40 Lakka systemd[1]: systemd-hostnamed.service: Succeeded.
Jul 20 05:08:10 Lakka sshd[509]: Accepted password for root from 192.168.0.11 port 58368 ssh2
Jul 20 05:11:20 Lakka systemd[1]: Stopping Retroarch...
Jul 20 05:11:20 Lakka systemd[1]: retroarch.service: Succeeded.
Jul 20 05:11:20 Lakka systemd[1]: Stopped Retroarch.
```

gouchi, 2021-07-20:
I presume you have the issue with `mupen64plus_next` and not `parallel-n64` ?

[Source](https://github.com/libretro/Lakka-LibreELEC/pull/1384)

gouchi, 2021-07-20:
Duplicate of #1332 if you are using `mupen64plus_next`.

closed 2021-07-26 by frani