libretro/Lakka-LibreELEC #1431: No HDMI signal after boot/logo screen on 4K TV

matheusgomes28, 2021-07-20:
- **Which version of Lakka are you using?**  
3.2 & Nightly Builds
- **What system hardware are you using?**
Raspberry Pi 4
- **What did you do?**
Plugged my RPI4 Lakka build to my living room TV (UHD 4K).
- **What did you expect to happen?**
To get HDMI signal
- **What happened instead?**
No HDMI signal after bootloader screen

The contents of my `config.txt` is currently the following:

```
# SPDX-License-Identifier: GPL-2.0-or-later                                                          
# Copyright (C) 2009-2014 Stephan Raue (stephan@openelec.tv)                                         
# Copyright (C) 2016-present Team LibreELEC (https://libreelec.tv)                                   
################################################################################                     
# Bootloader configuration - config.txt                                                              
################################################################################                     
                                                                                                     
################################################################################                     
# Memory (System/GPU configuration )                                                                 
################################################################################                     
                                                                                                     
# Default GPU memory split - at least 288M is needed for some 4k HEVC files                          
over_voltage=6                                                                                       
arm_freq=2147                                                                                        
gpu_freq=750                                                                                         
gpu_mem=384                                                                                          
                                                                                                     
################################################################################                     
# For overclocking and various other settings, see:                                                  
# https://www.raspberrypi.org/documentation/configuration/config-txt.md                              
################################################################################                     
# Set 'force_turbo=1' to disable dynamic overclocking and enable overclocking always.                
force_turbo=0                                                                                        
                                                                                                     
# Force HDMI even if unplugged or powered off                                                        
#hdmi_force_hotplug=1                                                                                
                                                                                                     
# Doesn't sent initial active source message.                                                        
# Avoids bringing CEC (enabled TV) out of standby and channel switch when                            
# rebooting.                                                                                         
#hdmi_ignore_cec_init=1                                                                              
#hdmi_drive=2                                                                                        
#hdmi_ignore_edid=0xa5000080                                                                         
#config_hdmi_boost=4                                                                                 
#hdmi_group=1                                                                                        
hdmi_safe=1                                                                                          
                                                                                                     
################################################################################                     
# End of default configuration                                                                       
################################################################################                     
                                                                                                     
# Choose between DVI and HDMI output                                                                 
# 1 = DVI mode (no sound)                                                                            
# 2 = HDMI mode (sound will be sent if supported and enabled)                                        
                                                                                                     
################################################################################                     
# Include distribution specific config file if it exists.                                            
################################################################################                     
[all]                                                                                                
include distroconfig.txt 
```
As you can see, I have previously turned the HDMI "safe" mode off and tried many combinations of the "hdmi_*" options. None of the solutions I looked up online have fixed this issues. I've looked at both RPI4 & Lakka forums.

Interestingly, HDMI output works fine on a slightly older DELL monitor I have in my office, but as soon as I plug it to my living room TV, I get no signal after the Lakka logo screen.

In addition, I have flashed both Manjaro and RaspberryPi OS to my Pi4. Both work very well on the new TV so I believe it's an issue with Lakka. Let me know if more information is needed here, I can probably get `dmesg` and other Lakka logs.

basilean, 2021-07-20:
Hello there,
 I can't found original reference regarding the issue but while it can sound non sense, adding: force_turbo=1 made trick for me.

matheusgomes28, 2021-07-20:
@basilean Unfortunately I have also tried that before, no luck :cry:

Ntemis, 2021-07-21:
try hdmi_enable_4kp60=0

matheusgomes28, 2021-07-22:
@Ntemis Apologies for the delay. I tried this right now with the following `/flash/config.txt`, and it didn't work either. 

```
# Default GPU memory split - at least 288M is needed for some 4k HEVC files
over_voltage=6
arm_freq=2147
gpu_freq=750
gpu_mem=384

################################################################################
# For overclocking and various other settings, see:
# https://www.raspberrypi.org/documentation/configuration/config-txt.md
################################################################################
# Set 'force_turbo=1' to disable dynamic overclocking and enable overclocking always.
force_turbo=1

# Force HDMI even if unplugged or powered off
hdmi_force_hotplug=1

# Doesn't sent initial active source message.
# Avoids bringing CEC (enabled TV) out of standby and channel switch when
# rebooting.
hdmi_ignore_cec_init=1
hdmi_drive=2
hdmi_enable_4kp60=0
#hdmi_ignore_edid=0xa5000080
#config_hdmi_boost=4
#hdmi_group=1
#hdmi_safe=1
```

Ntemis, 2021-07-23:
hdmi_max_pixel_freq=200000000 ;)

closed 2021-07-23 by Ntemis

matheusgomes28, 2021-07-23:
Added `hdmi_max_pixel_freq=200000000` to the `config.txt` above, still "no signal" after the bootloader screen :/

Ntemis, 2021-07-23:
Try latest nightly

Brunnis, 2021-07-23:
@Ntemis When I last tried the hdmi_max_pixel_freq setting on Lakka 3.2 a couple of weeks ago, it had no effect at all on my LG OLED65CX. Still ran happily in 4K. I ended up doing a custom start-up script that alters RetroArch's video_fullscreen_x/y settings to accomplish a similar behavior, as you might have seen on Discord. It would of course be nicer if the config.txt solution works, though.

Ntemis, 2021-07-23:
wait for the next nightly and test please

matheusgomes28, 2021-07-26:
Apologies for the delay, I'll do this test today! I was quite busy over the weekend :/

matheusgomes28, 2021-07-26:
I have just tested a new build, here are the main takeaways:

- The build works fine out of the box, HDMI signal was being detected by the TV without any `config.txt` hacks.
- When I overclocked with the same settings as the `config.txt` I posted earlier, the HDMI output fails. No signal detected by the TV.

So basically, it seems that Lakka doesn't properly setup the display settings when the RPI4 is overclocked, so this is the issue. Note that it only doesn't work when I'm using my 4K UHD TV, as it works just fine in my older monitor (when overclocked). In addition, other OS builds work just fine when overclocked on all displays I've tried.

So there must be something Lakka is doing differently to other builds, perhaps I'll look into it and see if I can help with anything.

In the meantime, has anyone else had this issue? Any fixes around? It would be really nice to be able to use my overclocked RPi4 to play some of the more demanding platforms :smile:

Ntemis, 2021-07-27:
You open an issue about you having no hdmi out signal and apparently this is fixed by your report
"The build works fine out of the box, HDMI signal was being detected by the TV without any config.txt hacks"
Overclocking your pi is another chapter you have to open over the pi forums not here.
Thank you for the confirmation that issue is fixed, have a nice gaming.

matheusgomes28, 2021-07-27:
@Ntemis I opened the issue because, well... it seemed like a bug. If you've ever worked in software development you shouldn't be annoyed that someone opened a bug that was fixed in a new build :laughing: , it's a pretty normal thing to do.

Regarding having to open issues with the RPi4 forums, I already did that. But to be honest with you, this is definitely not a Pi4 issue. As I said, HDMI on overclock works perfectly well with three other OS builds (Manjaro, Ubuntu, RPi OS). So there's something different about Lakka.

Anyway, I'll stop bothering this thread. Thanks for your suggestions and stay humble :+1:

Ntemis, 2021-07-27:
Last time i checked Lakka was not a general purpose operating system distro.
All those you mentioned above ((Manjaro, Ubuntu, RPi OS) don't push your hardware to its limits nor use the latest cutting etch in software development. Lakka is the most up to date operating system you can get your hands on atm.
Also note that every board is not made equal so some take maximum overclocking some dont.
We dont make the kernel nor the drivers for any system we support but we use what readily available out there.
But yes indeed there is something different about lakka! Lakka is made out of love of gaming and anything that comes out of love is humble by its nature.