libretro/Lakka-LibreELEC #1433: ParaLLEl N64 hle RSP Plugin broken + Mupen64plus-next broken

theAlphaFox, 2021-07-23:
On Lakka stable 3.2 on a Raspberry Pi 3B, the HLE RSP Plugin in ParaLLEl N64 doesn't allow the game to start, it just freezes at a black screen (I can still access the menu). CXD4 works, but it results in extremely poor performance compared to RetroPie. Mupen64plus-next does not run at all, it just crashes and brings me back to the menu.

RSP log dump (if of any use): 
[log-2021-07-23-02.30.21.zip](https://github.com/libretro/Lakka-LibreELEC/files/6866305/log-2021-07-23-02.30.21.zip)

Mupen64plus log dump: 
[log-2021-07-23-02.32.08.zip](https://github.com/libretro/Lakka-LibreELEC/files/6866307/log-2021-07-23-02.32.08.zip)

dmrlawson, 2021-07-24:
Mupen we know doesn't work atm. There's a potential driver issue I've been meaning to look at (see https://github.com/libretro/mupen64plus-libretro-nx/issues/357)

Parallel: I think we're not building very well. I'll see if I can add something to the makefile for this case.

gouchi, 2022-03-02:
@theAlphaFox Is it still an issue with [nightly builds](https://nightly.builds.lakka.tv/) ?

Thank you.

Ntemis, 2022-06-10:
this is stalled issue

closed 2022-06-10 by Ntemis