libretro/Lakka-LibreELEC #1442: Lakka 3.3 Crashing on Raspberry Pi 3

Rstation4, 2021-08-04:
Lakka 3.3 Stable Release (aarch 64) for Raspberry Pi 3
Raspberry Pi 3 B (1GB)

Tends to crash frequently, it will go back to the black Lakka splash screen before returning to the xmb menu. When it crashed on me the last 2 times I was in the XMB menu.

Settings: Custom Background used, Dynamic background (off), Horizontal Animation (off),  Menu Shader Pipeline (off), Menu Icon Theme (Retrosystem), Advance Settings On, Kiosk mode (off).

logs:
[retroarch.log](https://github.com/libretro/Lakka-LibreELEC/files/6927903/retroarch.log)

Ntemis, 2021-08-04:
Try to delete /storage/.config/retroarch/retroarch.cfg and try again

Ntemis, 2021-08-04:
Btw looks like we have an asset that doesnt get copied 
[WARN] [XMB] Asset missing: /tmp/assets/xmb/retrosystem/png/disc.png

Ntemis, 2021-08-04:
https://github.com/libretro/retroarch-assets/tree/master/xmb/retrosystem/png
There is no disc.png what the heck?

Rstation4, 2021-08-04:
I've tried it after deleting the retroarch.cfg file, and changing my theme to another. It still crashed and the 2nd time it froze. I had to power it off physically when it froze, I don't think the logs captured what had caused it, but I will add them anyway.

logs:
[Uploading retroarch.log…]()

dmrlawson, 2021-08-04:
> https://github.com/libretro/retroarch-assets/tree/master/xmb/retrosystem/png
> There is no disc.png what the heck?

Seems to exist for Ozone but not xmb, though it looks like it should:
https://github.com/libretro/RetroArch/commit/619fbd0e60bc299523d2b144dc4b08975686f0f2#diff-a416097f553e144f63fe6ce6929f922bb88c0239c309dd3101537516958af534R4960

@Rstation4 have you got any overclock (or other settings in config.txt/distroconfig.txt) different from default?

Rstation4, 2021-08-05:
No overclocking. I was just navigating through the xmb menu and it froze up on me again. Since I had deleted the cfg I see some of my changes were back to default.

Apart from changes and settings, I have the following: Kisok mode off, Advance settings on,XMB menu

Appearance settings: dynamic background tab on, Horizontal Animation (on), Custom background used, Menu Icon theme (systematic), Menu Shader Pipeline off.

Menu Item Visibility: Show Netplay (off), Show Date and Time (off), Show 'Explore' (off).
Misc: 5 system playlists with thumbnails

I'm going to turn Horizontal Animation and Dynamic Background back off.

dmrlawson, 2021-08-05:
Hm, I hadn't seen this before but could reproduce on an older version by trying to update to 3.3 using the Online Updater. When the download reached 96% or 99% it would freeze for over 5 minutes.

I suspect it might be running out of memory. I tried again and watched top, this is where it froze:

[image]

I changed `gpu_mem_1024=256` to `gpu_mem_1024=128` in `config.txt` and this allowed the Online Updater to finish successfully. There was a short freeze at the end but it did recover.

Could you please try changing this setting and report back?

Rstation4, 2021-08-07:
This is what is shown in config.txt.

_# Default GPU memory split - at least 288M is needed for some 4k HEVC files
gpu_mem=384_

RobLoach, 2021-12-09:
Just tested 3.6 on a RPi3 and didn't experience any freezes. Have you had any recurring issues on later Lakkas?

MessiahMozgus, 2021-12-09:
> Just tested 3.6 on a RPi3 and didn't experience any freezes. Have you had any recurring issues on later Lakkas?

A week ago it refused to update after 3 tries. It downloaded but never do anything with the update. Rebooting did nothing. I'm back on retropie now after finding out how Lanka broke RCA jack support months ago. I just got a Sony Trinitron so Lanka is no use to me now

Ntemis, 2021-12-09:
Version 3.3 is no more the current version and we didn't get any updates on this after newer ones have released

closed 2021-12-09 by Ntemis