libretro/Lakka-LibreELEC #1450: The controller only works on port 1

yikyo, 2021-08-10:
- Which version of Lakka are you using?
   - Lakka 3.3
- What system hardware are you using?
   - Intel NUC D54250 
- What did you do?
   - N/A
- What did you expect to happen?
   - both controller is working fine
- What happened instead?
   - just work on Port 1, Port 2 controller is not working

 I have two bluetooth controllers (8bitdo), Both use Bluetooth connection，
 But only works fine on port 1, When I change the device of port 1，The other device is working fine，

 At the same time, only the device on Port 1 is working properly，In XMB or games

 Sorry for my English

gouchi, 2021-08-10:
> It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at https://www.lakka.tv/doc/Troubleshooting-Lakka/

[Source issue template](https://github.com/libretro/Lakka-LibreELEC/blob/master/.github/ISSUE_TEMPLATE.md)

yikyo, 2021-08-10:
### System Log

[log.txt](https://github.com/libretro/Lakka-LibreELEC/files/6961799/log.txt)

yikyo, 2021-08-10:
### dmesg output

[log.txt](https://github.com/libretro/Lakka-LibreELEC/files/6961827/log.txt)

yikyo, 2021-08-10:
This time I use usb connection and still have the same problem

gouchi, 2021-08-10:
Thank you.

Can you check if you are using [latest firmware](https://support.8bitdo.com/firmware-updater.html) for your 8bitdo controllers ?

Which mode did you try ?

```[  358.446941] hid-generic 0003:2DC8:6002.0004: input,hidraw0: USB HID v1.11 Gamepad [8Bitdo SF30 Pro   8BitDo SN30 Pro+] on usb-0000:00:14.0-2/input0
[  375.022730] usb 2-1: new full-speed USB device number 7 using xhci_hcd
[  375.166787] input: 8Bitdo SF30 Pro   8Bitdo SN30 Pro as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/0003:2DC8:6001.0005/input/input15
[  375.167060] hid-generic 0003:2DC8:6001.0005: input,hidraw1: USB HID v1.11 Gamepad [8Bitdo SF30 Pro   8Bitdo SN30 Pro] on usb-0000:00:14.0-1/input0
```

yikyo, 2021-08-11:
windows and android mode，I will update the firmware later and test again

yikyo, 2021-08-12:
It still doesn't work after updating the firmware.
I tested 3 official image（3.0，3.3，nightly-20210808）, and none of them work properly
But in lakka2.1, the unofficial image, it works fine，
What settings are needed to control the menu at the same time
thank you
@gouchi

gouchi, 2021-08-13:
Thank you for the information.

Can you check if the controller is working using those commands in the terminal ?

List available controllers
`ls /dev/input/js*`

For controller 1, it should be
`jstest /dev/input/js0`

For controller 2, it should be
`jstest /dev/input/js1`

Also do you see a notification when you plugged your controller ?

Thank you.

yikyo, 2021-08-13:
I reinstalled the system and only enabled ssh

`ls /dev/input/js*`
[ls.log](https://github.com/libretro/Lakka-LibreELEC/files/6982905/ls.log)

`jstest /dev/input/js0`
[jstest.js0.log](https://github.com/libretro/Lakka-LibreELEC/files/6982914/jstest.js0.log)

`jstest /dev/input/js1`
[jstest.js1.log](https://github.com/libretro/Lakka-LibreELEC/files/6982931/jstest.js1.log)

I have tested all modes and none of them work

But when the port 2 controller is bound, it works properly

gouchi, 2021-09-02:
Hi,

According to the log the first one is Xbox controller which should be xinput mode from your 8Bitdo controller but the second one mentions `Sony Computer Entertainment Wireless Controller` so you are using at the end 8Bitdo and DS4 controllers for your test ? 

In Settings tab > Input > Port 2 Controls > Device type is set to `RetroPad` not `None` ?