libretro/Lakka-LibreELEC #1457: Acheievement Notification Crashes Audio

paqman3d, 2021-08-12:
Using Lakka 3.3 on the Asus Tinkerboard, I noticed whenever I unlock an achievement while using the Genesis Plus GX core, simply the pop up notification causes the sound to turn to static noise.

This worked fine last version. I have to quit the game and reboot each time now. Quick way to test is to load up Mortal Kombat 1 on genesis with achievements on. Enter the ABACABB code at the start and it should trigger a notification. 

I disabled the unlock sound but that didn't help. 

Also, Lakka tends to randomly crash on Tinkerboard on 3.3 but I'm not exactly sure what the issue is with that so that may be a different post.

BrianRabil, 2021-08-16:
I'm experiencing the same issue on a pi4

gouchi, 2021-08-18:
I can't reproduce your issue using nightly with Tinkerboard. When I unlock achievement using Genesis Plus GX core, the notification appears without sound issue.

@BrianRabil both arm and arm64 ?

Can you make a test with [nightly builds](https://nightly.builds.lakka.tv/latest/) if you reproduce the issue ?

Thank you.

paqman3d, 2021-08-29:
I tested MK1 again in Genesis Plus GX on a nightly from 8/17. The static still appears with the notification on Tinkerboard.

gouchi, 2021-08-30:
@paqman3d You did change some audio parameters perhaps ?

paqman3d, 2021-09-01:
> @paqman3d You did change some audio parameters perhaps ?

Nope. The settings were the same from when it worked prior to 3.2 I believe. Sound driver is on alsathread and the sound for Genesis Plus GX are on default values. 

I updated to the latest nightly and when I turn off the achievement sound it works fine. However when I go to the menu and come back, the static returns. Not even triggered by a notification, just the menu overlay. Using Ozone.

gouchi, 2021-09-01:
Not sure if [12921](https://github.com/libretro/RetroArch/pull/12921) will fix this issue.

Are you using perhaps  the sinc resampler with low or lowest quality ?

paqman3d, 2021-09-01:
> Not sure if [12921](https://github.com/libretro/RetroArch/pull/12921) will fix this issue.
> 
> Are you using perhaps the sinc resampler with low or lowest quality ?

I will have to check when I am home, but I believe I had it set on the lowest quality. I checked the linked issue and will change the setting and test again.

paqman3d, 2021-09-02:
> I will have to check when I am home, but I believe I had it set on the lowest quality. I checked the linked issue and will change the setting and test again.

I changed the audio quality to "normal" and so far it's acting correctly. Tested Mortal Kombat 1 and Ranger-X with no issues.

gouchi, 2021-09-02:
Thank you for the test.

May we close this issue as there is a workaround ?

And it will be fixed when update RA with next version after 1.9.8.

paqman3d, 2021-09-02:
> Thank you for the test.
> 
> May we close this issue as there is a workaround ?
> 
> And it will be fixed when update RA with next version after 1.9.8.

Yes I appreciate the help!

closed 2021-09-02 by gouchi