libretro/Lakka-LibreELEC #1458: Lakka 3.x - XBOX-360 compatible wireless controller detected as Nintendo Switch controller

muelli1975, 2021-08-13:
Hello,

I'm running Lakka on a Raspberry Pi 4 4 GB and my XBOX-360 compatible wireless controller is randomly detected as Nintendo Switch Controller when I restart or the device. The behavaviour occurs on all Lakka Versions since 3.0, including the latest nightly. The controller is from EasySMX and worked without problems with Lakka 2.3.2. If I start my RasPi, most of the time the controller is detected as XBOX-360 compatible. If the detection is wrong, reconnectiong the wireless dongles or restarting the pi in most cases leads to a correct detection, but sometimes I have to repeat the procedure.

Best regards,
Christoph

Ntemis, 2021-08-13:
lsusb and dmesg please

GavinDarkglider, 2021-08-14:
the pi images include joycond, which creates a virtual udev based controller. Try changing your retroarch controller settings, or stopping joycond.

gouchi, 2022-03-02:
@muelli1975 Is it still an issue or we can close this issue ?

closed 2022-06-10 by Ntemis