libretro/Lakka-LibreELEC #1465: Add these emulators to lakka on switch

DedoGamingONE, 2021-08-19:
Hello, is it possible for you guys to add these emulators to lakka on the switch?
- Citra
- Play!
- Duckstation

plaidman, 2021-08-19:
the latest nightly has duckstation. I have tried to compile Play and Citra in the past and it didn't take. Either way they wouldn't run anything faster than 30% speed.

DedoGamingONE, 2021-08-19:
Is this at the stock clock speed though? Because Im going to be running the switch at the maximum OC anyways.

plaidman, 2021-08-19:
Play and Citra have been compiled for lakka in the past (when the switch version was not merged with the upstream codebase) and they ran less than 20 fps with full overclock.

plaidman, 2021-08-19:
I don't want to put in the effort to make them work in the upstreamed lakka for that kind of performance.

DedoGamingONE, 2021-08-19:
Ok can you give me link to it so I could possibly add it to my own build?

Also I have citra on the HOS and on retroarch and it runs decently. I tested LOZ OOT and LOZ Tri Force heros. So I dont really know if that applies to all games however.

plaidman, 2021-08-19:
The HOS version of citra has had a LOT of blood sweat and tears put in by m4xw to make it run well on the switch. These changes are not available to the main libretro branch yet, and I don't know if those improvements would fully apply to the lakka build even when they are available.

plaidman, 2021-08-19:
[play_libretro.tar.gz](https://github.com/libretro/Lakka-LibreELEC/files/7018119/play_libretro.tar.gz)
You can try this file, it's the build that was compiled for the older version of lakka. No guarantees that it will work. unzip the .so file into `/lakka/storage/cores`

plaidman, 2021-08-19:
I'll try to pluck out the one for citra too, give me a moment

DedoGamingONE, 2021-08-19:
yeah i saw what m4xw has done and supported him on patreon for a while. Also thank you so much for this.

plaidman, 2021-08-19:
ah nevermind, I forgot Nati took down the old builds.

DedoGamingONE, 2021-08-19:
Ah damn. thanks anyways

DedoGamingONE, 2021-08-20:
> [play_libretro.tar.gz](https://github.com/libretro/Lakka-LibreELEC/files/7018119/play_libretro.tar.gz)
>  unzip the .so file into `/lakka/storage/cores`
And what about the .info file? Also it doesnt seem to work :( says "Failed to open libretro core". Sadge

gouchi, 2021-09-05:
@DedoGamingONE 

Duckstation and Play! core are available for arm64.

Play! core is still experimental but promising but some games are already [playable](https://github.com/jpd002/Play-Compatibility/issues?q=is%3Aopen+is%3Aissue+label%3Astate-playable+sort%3Aupdated-desc).

Can you make a test on [Lakka 3.4](https://le-builds.lakka.tv/Switch.aarch64/Lakka-Switch.aarch64-3.4.7z) ?

Thank you.

DedoGamingONE, 2021-09-06:
Ok I will test it out soon.

DedoGamingONE, 2021-09-06:
@gouchi So I can see that both duckstation and play! cores are available in the core section, but I cant launch a game with either of them. It just goes to a black screen for 2 seconds then back to the home screen.

Ntemis, 2021-09-06:
> @gouchi So I can see that both duckstation and play! cores are available in the core section, but I cant launch a game with either of them. It just goes to a black screen for 2 seconds then back to the home screen.

Bios

closed 2021-09-06 by Ntemis

gouchi, 2021-09-06:
For Play! you must try only with [playable games](https://github.com/jpd002/Play-Compatibility/issues?q=is%3Aopen+is%3Aissue+label%3Astate-playable+sort%3Aupdated-desc). And it if still not working you should provide some [logs](http://www.lakka.tv/doc/Troubleshooting-Lakka/). Thank you.

DedoGamingONE, 2021-09-06:
I see. Is it only limited to the playable games list?

gouchi, 2021-09-07:
Yes please read  [game statuses](https://github.com/jpd002/Play-Compatibility#game-statuses).

GavinDarkglider, 2021-09-07:
No one is going to include duckstation, since it is closed source. we have swanstation already, and a way to include the closed source duckstation core if you want that already. As for Citra, that core was always broken on the switch, and I opted out of including it. And play will require adding "Breaking fixes" to the video card driver..... they might have updated the vulkan API, but nvidia broke the video card falcon handling on jetson based platforms, which causes huge performance hits, so I will not be updating that anytime soon. On top of that, in order to update that, I would also have to port all of the switch kernel changes into a newer L4T kernel. And really, I am happy with lakka as it is. The only thing I might consider fixing is the decoder stuff, which is in a sort of working, sort of not working state.