libretro/Lakka-LibreELEC #1470: GCC 9.4 breaks Stella compilation

BrozzSama, 2021-08-24:
- Which version of Lakka are you using?
Latest commit from Github

- What system hardware are you using?
Compiling with target Raspberry Pi 1 from an x86_64 system.

- What did you do?
Followed the compilation instructions.

- What did you expect to happen?
Compilation should complete successfully.

- What happened instead?
The compilation fails when trying to build stella:
```
/home/francesco/Lakka-LibreELEC/build.Lakka-RPi.arm-3.3-devel/gcc-9.4.0/libstdc++-v3/src/c++98/istream.cc:116: error: undefined reference to 'std::istream::ignore(int)'
/home/francesco/Lakka-LibreELEC/build.Lakka-RPi.arm-3.3-devel/gcc-9.4.0/libstdc++-v3/src/c++98/istream.cc:358: error: undefined reference to 'std::basic_istream<wchar_t, std::char_traits<wchar_t> >::ignore(int)'
collect2: error: ld returned 1 exit status
make: *** [Makefile:630: stella_libretro.so] Error 1
```

The problem is resolved when downgrading the version to GCC 9.3 in the gcc's package.mk file.

The complete error log can be found [here](https://github.com/libretro/Lakka-LibreELEC/files/7042352/error.log)

gouchi, 2021-08-28:
Hi,

You have just changed the gcc version ? or revert to older commit ?

Because we added recently l[ibstdc++.a to sysroot](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/lang/gcc/package.mk#L89) recently.

Thank you.

BrozzSama, 2021-08-29:
> Hi,
> 
> You have just changed the gcc version ? or revert to older commit ?
> 
> Because we added recently l[ibstdc++.a to sysroot](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/lang/gcc/package.mk#L89).
> 
> Thank you.

Hi,

I wasn't really sure about whether I reverted or not, therefore I ran the compilation on the latest commit and disabled the line that copies libstdc++ to root. Apparently that is what's actually causing the error, I can investigate further but I don't really know what I'm looking for since I've never worked with LibreELEC/Lakka.

Ntemis, 2021-08-31:
how are you building the core?

BrozzSama, 2021-09-01:
> how are you building the core?

I simply followed the procedure outlined here: https://www.lakka.tv/doc/Compiling-Lakka/. If that line is added everything works up to the point where we have to build the stella core.

Ntemis, 2021-09-01:
What is the os you are using? Did you try pulling in the latest changes delete the current build directory and start clean?

BrozzSama, 2021-09-02:
I'm using Ubuntu 20.04 x86_64. I've cloned the repo in order to start clean. If [libstdc++ is added to sysroot](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/lang/gcc/package.mk#L89), the stella core does not compile and the build fails.

I'm attaching the complete log of the failing core.

[stderr.txt](https://github.com/libretro/Lakka-LibreELEC/files/7099785/stderr.txt)
[stdout.txt](https://github.com/libretro/Lakka-LibreELEC/files/7099786/stdout.txt)

gouchi, 2021-09-02:
This is not the best solution but more of a workaround : we could disable using  libgcc & libstdc++ static for the compilation 

```
diff --git a/Makefile b/tmp/Makefile
index a170992343..6a828521e5 100644
--- a/Makefile
+++ b/tmp/Makefile
@@ -57,11 +57,10 @@ ifneq ($(GIT_VERSION)," unknown")
    CXXFLAGS += -DGIT_VERSION=\"$(GIT_VERSION)\"
 endif
 
-GET_STATIC_ARG=$(if $(filter $(shell $(CXX) -print-file-name=$1.a),$1.a),,-static-$1)
 # Unix
 ifneq (,$(findstring unix,$(platform)))
    CXXFLAGS += $(LTO)
-   LDFLAGS += $(LTO) $(PTHREAD_FLAGS) $(call GET_STATIC_ARG,libgcc) $(call GET_STATIC_ARG,libstdc++)
+   LDFLAGS += $(LTO) $(PTHREAD_FLAGS)
    TARGET := $(TARGET_NAME)_libretro.so
    fpic := -fPIC
    ifneq ($(findstring SunOS,$(shell uname -a)),)
```

commit a415de50 (2021-09-03): stella (patch): do not use static libgcc/libstc++

fixes #1470
closed 2021-09-03 by ToKe79