libretro/Lakka-LibreELEC #1476: 'Lightgun' Setting Ignores Click Press Systemwide

mediamoshpit, 2021-08-27:
- Which version of Lakka are you using?
RPi4.arm-3.3

- What system hardware are you using?
Raspberry Pi 4B 2gb

- What did you do?
Tested mouse-based light gun (Aimtrak)

- What did you expect to happen?
'Mouse' and 'lightgun' setting to respect selected mouse port setting.

- What happened instead?
'Mouse' setting behaves as it should. 'Lightgun' respects only movement when mouse port is set to anything other than '0' and ignores mouse click entirely. This behavior is systemwide.

mediamoshpit, 2021-08-30:
Tested this a bit more today. It's strange; whenever the cursor is pointed offscreen, the mouse/trigger will register every time. Once the cursor is visible onscreen, input becomes wildly inconsistent. Turns out that every 10-20th click will register and if you hold it down, it will continue to register until released. Afterwards it will again only register one out of every 10-20 clicks.

gouchi, 2022-03-02:
Is it still an issue with [nightly builds](https://nightly.builds.lakka.tv/) ?

Thank you.

Ntemis, 2022-06-10:
Stalled issue

closed 2022-06-10 by Ntemis