libretro/Lakka-LibreELEC #1477: Retroarch.cfg is messed up when shutting down on generic x64

coldspark29, 2021-08-29:
- Which version of Lakka are you using?
    generic x64

- What system hardware are you using?
   ThinkCentre M93p

- What did you do?
  Shut down the system

- What did you expect to happen?
  That Lakka would only shutdown

- What happened instead?
  My retorach.cfg is messed up. The menu driver changed, all the playlists and settings are gone. Second time now.

Ntemis, 2021-08-29:
Did you save the config prior to shutdown?

coldspark29, 2021-08-30:
Not that I know of. It automatically saves the config by default when you close Retroarch, but even then it should be fine, because it was fine when I shutdown Lakka. When I delete the ``retroarch.cfg``, everything is fine again. So no idea why it suddenly sets the driver to Ozone. Even when you chosse xmb again, the fonts are messed up.

Ntemis, 2021-08-31:
Try running a memtest and next time save the retroarch.cfg prior you shutdown

coldspark29, 2021-09-01:
Yeah, I was also thinking my memory could be faulty. It is an old computer. Will try that.

coldspark29, 2021-10-17:
Hasn't happened in a while. Could have also been a memory issue, because I exchanged them. Closing.

closed 2021-10-17 by coldspark29