libretro/Lakka-LibreELEC #1478: After the first core selection, the cursor lands on "Information"

coldspark29, 2021-08-29:
- Which version of Lakka are you using?
  497be54

- What system hardware are you using?
  Raspberry Pi and x64

- What did you do?
  Select a core the first time I play a rom

- What did you expect to happen?
  That the cursor would stay at its position

- What happened instead?
  It visually moves to "Information", although pressing the button again does start the game and not open information.

Ntemis, 2021-08-29:
This belongs to retroarch issues not here, 
https://github.com/libretro/RetroArch/issues
Have fun

closed 2021-08-29 by Ntemis

coldspark29, 2021-08-30:
I don't have this issue on Retroarch on my PC. I think it is a Lakka bug or Retroarch should be updated.

Ntemis, 2021-08-30:
Your pc is on linux?

coldspark29, 2021-08-30:
Arch btw