libretro/Lakka-LibreELEC #1479: Missing Hotkeys

SkushaKutusha, 2021-08-30:
Why are there no hotkeys to enter the lakka menu during the game? In 2.3.2 they worked (select + x, start + select), but in 3.3 - they are missing, i.e. you can not go to the menu, the console with a gamepad does not reload. And how to turn off the pop-up window with the name of the rom in the center of the screen when starting the game?

admiralspeedy, 2021-08-31:
The option to set them is there for me, however the menu hotkey does not work for me no matter what I set it to with a USB SNES controller clone. The other hotkeys work when I set them, just the menu doesn't.

gouchi, 2021-09-02:
If I am not mistaken, there are two ways to display the menu during playing 

Menu toggle :  `Settings tab > Input > Hotkeys > Menu (Toggle)` which should be link to your joypad config.

Menu toggle Controller Combo :  `Settings tab > Input > Hotkeys > Menu toggle Controller Combo` and we set by default to `L3+R3`

> And how to turn off the pop-up window with the name of the rom in the center of the screen when starting the game?

Settings tab > On Screen Display > On Screen Notifications > Notification Visibility > "Load Content" Startup Notification

Please bear in mind as written in the [github issue template](https://github.com/libretro/Lakka-LibreELEC/blob/master/.github/ISSUE_TEMPLATE.md)
```
This issue tracker is NOT for help or support. Please use one of the following channels for help and support:

the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general)
via IRC @ #lakkatv on freenode.net
#lakkatv on libretro discord https://discord.gg/retroarch
```

Please reopen the issue if needed.

closed 2021-09-02 by gouchi