libretro/Lakka-LibreELEC #1480: linuxraw Input Driver Bootloop

admiralspeedy, 2021-08-31:
Running Lakka 3.3 ARM on a Raspberry Pi 4, 4GB.

I set my input driver to linuxraw, to see if it would fix the menu hotkey not working (I changed it to Start + Select but I cannot get the menu to open in any core) and after rebooting the device, Lakka now refuses to boot and simply flickers the logo on screen every second or two.

dmrlawson, 2021-09-01:
> Running Lakka 3.3 ARM on a Raspberry Pi 4, 4GB.
>
> I set my input driver to linuxraw, to see if it would fix the menu hotkey not working (I changed it to Start + Select but I cannot get the menu to [...]

Hi,

I sounds like retroarch is failing to start. Please could you provide some logs from retroarch?

Instructions here: https://www.lakka.tv/doc/Troubleshooting-Lakka/

gouchi, 2022-03-02:
Closing this issue as we did not receive any logs.