libretro/Lakka-LibreELEC #1481: Current MAME core on Lakka Switch or Any Platform.

vus7294, 2021-09-01:
https://github.com/libretro/mame
Here is the Project Link. Current MAME core is on Linux and android. so It might be easy to Port Lakka core. What do you Think about this?
and Nothing to Working Game in 2015 core. I Tried CV1K games and Black Screen.

gouchi, 2021-09-02:
We already provide [libretro-mame core](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/libretro/mame/package.mk) and we did not [disable it per device](https://github.com/libretro/Lakka-LibreELEC/blob/master/distributions/Lakka/options#L454-L510).

So I don't understand your issue. Maybe I misunderstood ?

Thank you.

vus7294, 2021-09-06:
I tried Lakka Latest build but No MAME(Not 2003, 2015) current core is Present.
MAME Current core is already build arm64 core, so it might be easy to port switch.
MAME or MAMEARCADE is core name.

ToKe79, 2021-09-06:
@gouchi we do not build the core, it is not in the [list](https://github.com/libretro/Lakka-LibreELEC/blob/master/distributions/Lakka/options#L229-L231).

ToKe79, 2021-09-06:
I tried to build it for Switch - it uses the target toolchain to build a binary (`genie`), that is needed on the host and than wants to use that binary to build:
```
3rdparty/genie/bin/linux/genie --distro=debian-stable --OPTIMIZE=3 --NOWERROR='1' --target='mame' --subtarget='arcade' --build-dir='build' --PYTHON_EXECUTABLE='python2' --osd='retro' --targetos='linux' --PLATFORM='aarch64' --gcc=linux-gcc --gcc_version=9.4.0 gmake
3rdparty/genie/bin/linux/genie: 1: Syntax error: "(" unexpected
make: *** [makefile:1438: build/projects/retro/mamearcade/gmake-linux/Makefile] Error 2
make: *** Waiting for unfinished jobs....
```

```
$ file build.Lakka-Switch.aarch64/mame-7b6d18f/3rdparty/genie/bin/linux/genie
build.Lakka-Switch.aarch64/mame-7b6d18f/3rdparty/genie/bin/linux/genie: ELF 64-bit LSB executable, ARM aarch64, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-aarch64.so.1, for GNU/Linux 3.7.0, stripped
```
The host is x86_64, but the binary is for aarch64... we have to first build genie with host-toolchain and then continue the build for target. I personally do not know how to proceed from here, but this is my "finding".

GavinDarkglider, 2021-09-06:
I am not sure I am understanding the issue here.....When you say latest lakka which version did you download..... I have been known to build debug builds without the mame cores, because they take longer to build than just about anything else. lol. I think the current version of lakka is 3.3, and I know it has mame cores. That being said, I didnt test any of them, as it was never my focus to fix cores, or core issues on the switch. I was focused more on OS porting than the retroarch stuff. That being said, it appears you are trying to add a package that depends on another package. so create a package called genie, and build genie for host there, and make this package depend on it. Or if you get really bored, build the core in L4T-ubuntu, and copy the lib to the cores folder for lakka, which should get you around this limitation.

ToKe79, 2022-03-23:
You can test mame "current" with images here: http://nightly.builds.lakka.tv/PRs/pr_1650_000486/

ToKe79, 2022-03-24:
@vus7294 the core can be downloaded from [here](https://nightly.builds.lakka.tv/members/vudiq/mame/), then store it in `/tmp/cores` / samba share `Cores` / copy to `LAKKA_DISK:/cores`.

PR #1650 (2022-03-24): mame: enable mame current libretro core
does build also for Pi / Pi Zero, but not including in image
closes: #1481
closed 2022-03-24 by ToKe79