libretro/Lakka-LibreELEC #1484: Retroarch sometimes freezes for about a minute

coldspark29, 2021-09-04:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using? 
   56182da

- What system hardware are you using? 
   x64

- What did you do?
   Enter the menu while playing

- What did you expect to happen?
   Nothing really

- What happened instead?
   System froze for about a minute

I have experienced this a few times now since being on x64. Sometimes Retroarch freezes, you can't use any controls and then it goes back to normal. Luckily this doesn't happen while being in the game.

```
[INFO] RetroArch 1.9.8 (Git fa5ec2d4b1)
[INFO] === Build =======================================
[INFO] CPU Model Name: Intel(R) Core(TM) i5-4570T CPU @ 2.90GHz
[INFO] Fähigkeiten:  MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES AVX
[INFO] Built: Sep  4 2021
[INFO] Version: 1.9.8
[INFO] Git: fa5ec2d4b1
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Core]: Loading dynamic libretro core from: "/tmp/cores/mupen64plus_next_libretro.so"
[INFO] [Overrides]: No core-specific overrides found at "/storage/.config/retroarch/config/Mupen64Plus-Next/Mupen64Plus-Next.cfg".
[INFO] [Overrides]: No content-dir-specific overrides found at "/storage/.config/retroarch/config/Mupen64Plus-Next/n64.cfg".
[INFO] [Overrides]: No game-specific overrides found at "/storage/.config/retroarch/config/Mupen64Plus-Next/Banjo-Kazooie (Europe) (En,Fr,De).cfg".
[INFO] [Environ]: GET_CORE_OPTIONS_VERSION.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_V2.
[INFO] [Environ]: SET_CONTROLLER_INFO.
[INFO] [Remaps]: Remap directory: "/storage/remappings".
[INFO] [Overrides]: Speicherdaten umleiten in "/storage/savefiles/Banjo-Kazooie (Europe) (En,Fr,De).srm".
[INFO] [Overrides]: Spielstand umleiten in "/storage/savestates/Banjo-Kazooie (Europe) (En,Fr,De).state".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Environ]: GET_PERF_INTERFACE.
[INFO] [Environ]: SET_PIXEL_FORMAT: XRGB8888.
[INFO] [Environ]: GET_RUMBLE_INTERFACE.
[INFO] [CONTENT LOAD]: Lade Inhalts-Datei: /storage/roms/sdb1-usb-WDC_WDS_100T2B0A/Games/ROMs/n64/Banjo-Kazooie (Europe) (En,Fr,De).n64
[INFO] Kein gültiger Patch für diesen Inhalt gefunden.
[INFO] [Environ]: SET_HW_RENDER, context type: glcore.
[INFO] Requesting core OpenGL context (3.3).
[INFO] Reached end of SET_HW_RENDER.
[libretro INFO] mupen64plus: Using full mem base
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[libretro INFO] mupen64plus: Goodname: Banjo-Kazooie (E) (M3) [!]
[libretro INFO] mupen64plus: Name: Banjo-Kazooie       
[libretro INFO] mupen64plus: MD5: 06A43BACF5C0687F596DF9B018CA6D7F
[libretro INFO] mupen64plus: CRC: 733FCCB1 444892F9
[libretro INFO] mupen64plus: Imagetype: .v64 (byteswapped)
[libretro INFO] mupen64plus: Rom size: 16777216 bytes (or 16 Mb or 128 Megabits)
[libretro INFO] mupen64plus: ClockRate = F
[libretro INFO] mupen64plus: Version: 1448
[libretro INFO] mupen64plus: Manufacturer: Nintendo
[libretro INFO] mupen64plus: Cartridge_ID: 4B42
[libretro INFO] mupen64plus: Country: Europe (0x50)
[libretro INFO] mupen64plus: PC = 80100400
[libretro INFO] mupen64plus: Save type: 0
[INFO] [SRAM]: Überspringe Laden des SRAM..
[INFO] Version der libretro-API: 1
[INFO] Kompiliert gegen API: 1
[INFO] [Audio]: Set audio input rate to: 44100.00 Hz.
[INFO] [Video]: Timings deviate too much. Will not adjust. (Display = 60.00 Hz, Game = 50.00 Hz)
[INFO] [Video]: Video @ fullscreen
[... 84 lines cut ...]
[INFO] [Playlist]: Lade Verlaufs-Datei: [/storage/.config/retroarch/content_image_history.lpl].
[INFO] [Playlist]: Favoritendatei wird geladen: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] [Playlist]: Written to playlist file: /storage/.config/retroarch/content_history.lpl
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[libretro INFO] mupen64plus: Game controller 0 (Standard controller) has a Rumble pak plugged in
[libretro INFO] mupen64plus: Game controller 1 (Standard controller) has a Rumble pak plugged in
[libretro INFO] mupen64plus: Game controller 2 (Standard controller) has a Rumble pak plugged in
[libretro INFO] mupen64plus: Game controller 3 (Standard controller) has a Memory pak plugged in
[libretro INFO] mupen64plus: Using CIC type X103
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [KMS]: New FB: 1920x1200 (stride: 7680).
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [KMS]: New FB: 1920x1200 (stride: 7680).
[libretro INFO] mupen64plus: Initializing 4 RDRAM modules for a total of 8 MB
[libretro INFO] mupen64plus: Starting R4300 emulator: Dynamic Recompiler
[libretro INFO] mupen64plus: Init new dynarec
[INFO] [udev]: Pad #0 (/dev/input/event3) supports force feedback.
[INFO] [udev]: Pad #0 (/dev/input/event3) supports 16 force feedback effects.

```

RobLoach, 2021-12-09:
I remember this happening too, but I believe it was fixed. Are you still experiencing it?

gouchi, 2022-03-02:
No answer closing.

closed 2022-03-02 by gouchi