libretro/Lakka-LibreELEC #1492: Boot error cubieboard 2, Lakka 3.4

johndoe71rus, 2021-09-07:
Lakka 3.4 release and Lakka-H3.arm-nightly-20210907-b753512-cubieboard2 
cubieboard 2
try to boot new release

> [21:04:55:295] U-Boot SPL 2019.04 (Sep 07 2021 - 00:35:49 +0200)
[21:04:55:295] DRAM: 1024 MiB
[21:04:55:295] CPU: 912000000Hz, AXI/AHB/APB: 3/2/2
[21:04:55:295] Trying to boot from MMC1
[21:04:55:415]
[21:04:55:420]
[21:04:55:420] U-Boot 2019.04 (Sep 07 2021 - 00:35:49 +0200) Allwinner Technology
[21:04:55:429]
[21:04:55:429] CPU:   Allwinner A20 (SUN7I)
[21:04:55:429] Model: Cubietech Cubieboard2
[21:04:55:429] I2C:   ready
[21:04:55:450] DRAM:  1 GiB
[21:04:55:970] MMC:   SUNXI SD/MMC: 0
[21:04:55:972] Loading Environment from FAT... *** Warning - bad CRC, using default environment
[21:04:56:000]
[21:04:56:523] In:    serial
[21:04:56:525] Out:   serial
[21:04:56:539] Err:   serial
[21:04:56:539] SCSI:  SATA link 0 timeout.
[21:04:56:757] AHCI 0001.0100 32 slots 1 ports 3 Gbps 0x1 impl SATA mode
[21:04:56:768] flags: ncq stag pm led clo only pmp pio slum part ccc apst
[21:04:56:768]
[21:04:56:768] Net:
[21:04:56:791] Error: ethernet@1c50000 address not set.
[21:04:56:791] eth-1: ethernet@1c50000
[21:04:56:800] starting USB...
[21:04:56:800] No controllers found
[21:04:56:800] Hit any key to stop autoboot:  0
[21:04:56:815] switch to partitions #0, OK
[21:04:56:825] mmc0 is current device
[21:04:56:853] Scanning mmc 0:1...
[21:04:56:858] Found /extlinux/extlinux.conf
[21:04:56:872] Retrieving file: /extlinux/extlinux.conf
[21:04:56:872] 177 bytes read in 6 ms (28.3 KiB/s)
[21:04:56:881] 1:⇥	Lakka
[21:04:56:881] Retrieving file: /KERNEL
[21:04:57:347] 8522440 bytes read in 469 ms (17.3 MiB/s)
[21:04:57:352] append: boot=UUID=0709-2954 disk=UUID=8cf04496-4655-42d6-a23e-198aa41f6ea1 quiet console=ttyS0,115200 console=tty1
[21:04:57:364] Retrieving file: /sun7i-a20-cubieboard2.dtb
[21:04:57:373] Skipping Lakka for failure retrieving fdt
[21:04:57:373] SCRIPT FAILED: continuing...
[21:04:57:399] scanning bus for devices...
[21:04:57:403] Found 0 device(s).
[21:04:57:416]
[21:04:57:416] Device 0: not available
[21:04:57:416] starting USB...
[21:04:57:416] No controllers found
[21:04:57:416] USB is stopped. Please issue 'usb start' first.
[21:04:57:416] starting USB...
[21:04:57:416] No controllers found
[21:04:57:436]
[21:04:57:436] Error: ethernet@1c50000 address not set.
[21:04:57:458]
[21:04:57:465] Error: ethernet@1c50000 address not set.
[21:04:57:481]
[21:04:57:487] Error: ethernet@1c50000 address not set.
[21:04:57:503]
[21:04:57:509] Error: ethernet@1c50000 address not set.
[21:04:57:509] No ethernet found.
[21:04:57:528]
[21:04:57:541] Error: ethernet@1c50000 address not set.
[21:04:57:541] missing environment variable: pxeuuid
[21:04:57:558]
[21:04:57:558] Error: ethernet@1c50000 address not set.
[21:04:57:558] Retrieving file: /extlinux/pxelinux.cfg/00000000
[21:04:57:583]

How do I boot the latest Lakka at cubieboard2? If you drop support this old garbage HW, why are you continuing to build releases and nightly images giving a hope?

ToKe79, 2021-09-13:
We build the releases, so maybe someone in the community will send fixes. I do not think someone from the devs has this device.

ToKe79, 2021-09-13:
can you check, that you have the path to the dtb file correct? I am not sure if the file is in the root folder by default:
> `[21:04:57:364] Retrieving file: /sun7i-a20-cubieboard2.dtb`

johndoe71rus, 2021-09-13:
according answers https://github.com/libretro/Lakka-LibreELEC/issues/1092 https://github.com/libretro/Lakka-LibreELEC/issues/1077 Allwinner A20 not supported many time ago. But you still build nightly and releases images.
I'll check the files in the evening.

johndoe71rus, 2021-09-13:
Drive LAKKA
extlinux/extlinux.conf
KERNEL
KERNEL.md5
SYSTEM
SYSTEM.md5

there is no sun7i-a20-cubieboard2.dtb as a year ago.

ToKe79, 2021-09-13:
you are right, the a20 boards are not supported anymore in LibreELEC 9.2 (which is the base for Lakka 3.x). I do not have anything else I could give you. we will remove the download links from lakka.tv to avoid future confusion.

commit ac76cd31 (2021-09-13): Allwinner: drop unsupported boards

closes #1492
closed 2021-09-13 by ToKe79