libretro/Lakka-LibreELEC #1494: Kor font Crash on Stable 3.4 Switch

vus7294, 2021-09-08:
[image]
Lakka 3.4 Stable on Switch. Default Video Setting on Vulkan.

gouchi, 2021-09-08:
Hi,

You did restart one time RetroArch ?

Do you reproduce the issue using `gl` or `glcore` video driver ?

Now RA should take automatically the font in  `/usr/share/retroarch-assets/pkg/korean-fallback-font.ttf`

Thank you.

vus7294, 2021-09-08:
I set language and Reset, then That Glitch.

I Always use Video driver as Vulkan.

gouchi, 2021-09-08:
When you say Reset you mean Restart RetroArch ?

I will have to make some test with Korean language.

losernator, 2021-09-14:
> When you say Reset you mean Restart RetroArch ?
> 
> I will have to make some test with Korean language.

CJK fallback Fonts feature doesn't seem to work properly.
(not just Switch version also happens in unix-distro like raspberry pi, odroid)

Actually Korean does not need fall-back feature. default XMB and OZONE already have enough font glyph for Korean language

gouchi, 2021-09-14:
I made some test on my PC with Xorg and using Korean language 

1. XMB or Ozone

Using video driver `gl` or `glcore` it is using font `assets/pkg/korean-fallback-font.ttf`
Using video driver `vulkan` it is using font `assets/ozone/regular.ttf`

I will have to make some test with Lakka.

vus7294, 2021-10-09:
> I made some test on my PC with Xorg and using Korean language 
> 
> 
> 
> 1. XMB or Ozone
> 
> 
> 
> Using video driver `gl` or `glcore` it is using font `assets/pkg/korean-fallback-font.ttf`
> 
> Using video driver `vulkan` it is using font `assets/ozone/regular.ttf`
> 
> 
> 
> I will have to make some test with Lakka.

Sadly This bug is STILL PRESENT in 3.5 Stable for Switch.

gouchi, 2021-11-06:
Can you try to make a test by switching to xmb menu ?

It seems if you set the xmb menu font to ` /usr/share/retroarch-assets/ozone/regular.ttf` it will work after restart RA (Settings tab > User Interface > Appearance > Menu Font)

You will have to enable advanced settings to get access to xmb menu font (Settings tab > User Interface > Show Advanced Settings).

Thank you.

vus7294, 2021-11-12:
It Works, But I Can't Set Language with OZONE UI. This Seems Bad Glitch. Thanks.

vus7294, 2022-04-14:
I Tested x64 Lakka Even Can't Display Korean on XMB Theme. :(

Ntemis, 2022-06-10:
take it to retroarch gh

closed 2022-06-10 by Ntemis