libretro/Lakka-LibreELEC #1495: Gamecon module does not work in Lakka 3.3 and the latest nightly version

BDJoe360, 2021-09-09:
When I use the command: modprobe gamecon map = 0,7,0,0,7,0; After the driver is loaded, the psx handle model that should have appeared in input > port 1 controls > device index now only displays disabled after modification!

gouchi, 2021-09-09:
Please provide some [log](http://lakka.tv/doc/Troubleshooting-Lakka/) as mentioned in the [template issue](https://github.com/libretro/Lakka-LibreELEC/blob/master/.github/ISSUE_TEMPLATE.md).

> It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at https://www.lakka.tv/doc/Troubleshooting-Lakka/

BDJoe360, 2021-09-09:
Log here:
1.
[INFO] [Config]: Loading default config.
[INFO] [Config]: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
[INFO] RetroArch 1.9.9 (Git 36888612eb)
[INFO] === Build =======================================
[INFO] CPU Model Name: Intel(R) Xeon(R) CPU           L3110  @ 3.00GHz
[INFO] 容量:  MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4
[INFO] Built: Sep  8 2021
[INFO] Version: 1.9.9
[INFO] Git: 36888612eb
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] libretro API 版本: 1
[INFO] 未按照 API 编译: 1
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [DRM]: Found 3 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 26 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Connector 2 connected: no
[INFO] [DRM]: Connector 2 has 0 modes.
[INFO] [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 1: (1920x1080) 1920 x 1080, 59.940201 Hz
[INFO] [DRM]: Mode 2: (1920x1080i) 1920 x 1080, 30.000000 Hz
[INFO] [DRM]: Mode 3: (1920x1080i) 1920 x 1080, 29.970100 Hz
[INFO] [DRM]: Mode 4: (1920x1080) 1920 x 1080, 50.000000 Hz
[INFO] [DRM]: Mode 5: (1920x1080i) 1920 x 1080, 25.000000 Hz
[INFO] [DRM]: Mode 6: (1920x1080) 1920 x 1080, 30.000000 Hz
[INFO] [DRM]: Mode 7: (1920x1080) 1920 x 1080, 29.970100 Hz
[INFO] [DRM]: Mode 8: (1920x1080) 1920 x 1080, 24.000000 Hz
[INFO] [DRM]: Mode 9: (1920x1080) 1920 x 1080, 23.976080 Hz
[INFO] [DRM]: Mode 10: (1600x900) 1600 x 900, 60.000000 Hz
[INFO] [DRM]: Mode 11: (1280x1024) 1280 x 1024, 60.019741 Hz
[INFO] [DRM]: Mode 12: (1280x720) 1280 x 720, 60.000000 Hz
[INFO] [DRM]: Mode 13: (1280x720) 1280 x 720, 59.940201 Hz
[INFO] [DRM]: Mode 14: (1280x720) 1280 x 720, 50.000000 Hz
[INFO] [DRM]: Mode 15: (1280x720) 1280 x 720, 30.000000 Hz
[INFO] [DRM]: Mode 16: (1280x720) 1280 x 720, 29.970100 Hz
[INFO] [DRM]: Mode 17: (1280x720) 1280 x 720, 24.000000 Hz
[INFO] [DRM]: Mode 18: (1280x720) 1280 x 720, 23.976162 Hz
[INFO] [DRM]: Mode 19: (1024x768) 1024 x 768, 60.003841 Hz
[INFO] [DRM]: Mode 20: (800x600) 800 x 600, 60.316540 Hz
[INFO] [DRM]: Mode 21: (720x576) 720 x 576, 50.000000 Hz
[INFO] [DRM]: Mode 22: (720x480) 720 x 480, 60.000000 Hz
[INFO] [DRM]: Mode 23: (720x480) 720 x 480, 59.940060 Hz
[INFO] [DRM]: Mode 24: (640x480) 640 x 480, 60.000000 Hz
[INFO] [DRM]: Mode 25: (640x480) 640 x 480, 59.940475 Hz
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
[INFO] [GL]: Found GL context: kms
[INFO] [GL]: Detecting screen resolution 1920x1080.
[INFO] [EGL] Found EGL_EXT_platform_base, trying eglGetPlatformDisplayEXT
[INFO] [EGL]: EGL version: 1.5
[INFO] [EGL]: Current context: 0x7fc93c006470.
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [GL]: Vendor: nouveau, Renderer: NV106.
[INFO] [GL]: Version: 4.3 (Compatibility Profile) Mesa 21.2.1.
[INFO] [GL]: Using resolution 1920x1080
[... 183 lines cut ...]
[INFO] [udev]: Keyboard #1: "Power Button" (/dev/input/event0).
[INFO] [udev]: Keyboard #2: "SIGMACHIP USB Keyboard" (/dev/input/event2).
[INFO] [udev]: Keyboard #3: "SIGMACHIP USB Keyboard Consumer Control" (/dev/input/event3).
[INFO] [udev]: Keyboard #4: "SIGMACHIP USB Keyboard System Control" (/dev/input/event4).
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [Video]: Found display server: null
[WARN] [udev]: Mouse grab/ungrab feature unavailable.
[INFO] ALSA: Using signed 16-bit format.
[INFO] ALSA: Period size: 384 frames
[INFO] ALSA: Buffer size: 1536 frames
wifi is not available
[INFO] [Display]: Found display driver: "gl".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
function is no-op
[INFO] [SRAM]: 游戏存档不会被保存。
[INFO] [Playlist]: 正在加载收藏文件: [/storage/.config/retroarch/content_favorites.lpl].
function is no-op
function is no-op
function is no-op
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).

BDJoe360, 2021-09-09:
> ###########################################
> # Lakka - The DIY retro emulation console #
> # ...... visit http://www.lakka.tv ...... #
> ###########################################
>
> Lakka (official): devel-20210908010416-fc2b556 (Generic.x86_64)
> [1;32mLakka[1;32m:[1;34m~ [0m# lsmod
> Module                  Size  Used by
> snd_emu10k1_synth      24576  0
> snd_emux_synth         40960  1 snd_emu10k1_synth
> snd_seq_midi_emul      20480  1 snd_emux_synth
> snd_seq_virmidi        16384  1 snd_emux_synth
> snd_seq_midi           20480  0
> snd_seq_midi_event     16384  2 snd_seq_midi,snd_seq_virmidi
> snd_seq                65536  6 snd_seq_midi,snd_seq_midi_event,snd_seq_virmidi,snd_seq_midi_emul,snd_emux_synth
> gamecon                24576  0
> parport                32768  1 gamecon
> 8021q                  24576  0
> rfkill                 20480  1
> snd_hda_codec_hdmi     57344  1
> snd_emu10k1           155648  1 snd_emu10k1_synth
> snd_hda_intel          45056  0
> snd_util_mem           16384  2 snd_emu10k1,snd_emux_synth
> snd_intel_dspcfg       20480  1 snd_hda_intel
> snd_ac97_codec        126976  1 snd_emu10k1
> [... 696 lines cut ...]
> [   13.176592] 8021q: 802.1Q VLAN Support v1.8
> [   18.154710] CIFS: Attempting to mount \\192.168.1.200\roms
> [   18.154734] CIFS: No dialect specified on mount. Default has changed to a more secure dialect, SMB2.1 or later (e.g. SMB3.1.1), from CIFS (SMB1). To use the less secure SMB1 dialect to access old servers which do not support SMB3.1.1 (or even SMB3 or SMB2.1) specify vers=1.0 on mount.
> [   20.965377] snd_hda_intel 0000:01:00.1: IRQ timing workaround is activated for card #1. Suggest a bigger bdl_pos_adj.
> [1;32mLakka[1;32m:[1;34m~ [0m# exit

log:
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2021.09.09 18:13:27 =~=~=~=~=~=~=~=~=~=~=~=
login as: root

gouchi, 2021-09-10:
@BDJoe360 Thank you.

@ToKe79 We may need to add those [kernel modules](https://github.com/libretro/Lakka-LibreELEC/pull/292/files) as I wrote 

> Needed for CONFIG_JOYSTICK_GAMECON

ToKe79, 2021-09-10:
@gouchi I see, I only enabled `CONFIG_PARPORT`. will push fix asap.

ToKe79, 2021-09-10:
@BDJoe360 please, test with next nightly (2021-09-11 and later) after it is built. if the issue is still present, please comment here/reopen.

BDJoe360, 2021-09-11:
OK, I've tried lakka-generic.x86_ 64-nightly-20210911。 Moreover, the psx handle can be set and used normally, but there is still a problem.
The problem is that the device index still displays disabled, and the psx controller is not displayed. I hope this problem can be corrected.
Finally, upload the Log of lakka-generic.x86 _ 64-nightly-2020-08-08_&nbsp; legacy I normally used before&nbsp;and The log of&nbsp; lakka-generic.x86_&nbsp; 64-nightly-2021-09-11 is expected to help solve this problem.

Log of 2020-08-08:

login as: Error writing session log (raw mode) to file: putty.log
root
***@***.***'s password:
###########################################
# Lakka - The DIY retro emulation console #
# ...... visit http://www.lakka.tv ...... #
###########################################

Lakka (official): devel-20200808003424-r28355-gaf3cd2648f (Generic.x86_64)
Lakka:~ # lsmod
Module&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; Size&nbsp; Used by
gamecon&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; 20480&nbsp; 0
8021q&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; 24576&nbsp; 0
snd_hda_codec_hdmi&nbsp; &nbsp; &nbsp;45056&nbsp; 1
snd_emu10k1&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp;143360&nbsp; 0
snd_util_mem&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp;16384&nbsp; 1 snd_emu10k1
snd_ac97_codec&nbsp; &nbsp; &nbsp; &nbsp; 110592&nbsp; 1 snd_emu10k1
ac97_bus&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp;16384&nbsp; 1 snd_ac97_codec
snd_rawmidi&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; 24576&nbsp; 1 snd_emu10k1
rng_core&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp;16384&nbsp; 0
snd_hda_intel&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; 28672&nbsp; 2
snd_hda_codec&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; 90112&nbsp; 2 snd_hda_intel,snd_hda_codec_hdmi
snd_hda_core&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp;49152&nbsp; 3 snd_hda_intel,snd_hda_codec,snd_hda_codec_hdmi
snd_hwdep&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; 16384&nbsp; 2 snd_emu10k1,snd_hda_codec
parport_pc&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp;36864&nbsp; 1
parport&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; 36864&nbsp; 2 gamecon,parport_pc
asus_atk0110&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp;20480&nbsp; 0
overlay&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; 49152&nbsp; 7
applespi&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp;28672&nbsp; 0
Lakka:~ # dmesg
[&nbsp; &nbsp; 0.000000] microcode: microcode updated early to revision 0xa0b, date = 2010&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; -09-28
[&nbsp; &nbsp; 0.000000] Linux version 4.11.12-rt14 ***@***.***) (gcc version 9.2.0 (G&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; CC) ) #1 SMP PREEMPT RT Tue Jan 21 03:44:14 CET 2020
[&nbsp; &nbsp; 0.000000] Command line: BOOT_IMAGE=/KERNEL boot=UUID=0808-0218 disk=UUID=1a&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; a4db76-ef24-48d3-9052-3dff684a02fb portable quiet vt.global_cursor_default=0 log&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; level=2
[&nbsp; &nbsp; 0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point regi&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; sters'
[&nbsp; &nbsp; 0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[&nbsp; &nbsp; 0.000000] x86/fpu: Enabled xstate features 0x3, context size is 576 bytes,&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp;using 'standard' format.
[&nbsp; &nbsp; 0.000000] e820: BIOS-provided physical RAM map:
[&nbsp; &nbsp; 0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[&nbsp; &nbsp; 0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[&nbsp; &nbsp; 0.000000] BIOS-e820: [mem 0x00000000000e4000-0x00000000000fffff] reserved
[&nbsp; &nbsp; 0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007ff8ffff] usable
[&nbsp; &nbsp; 0.000000] BIOS-e820: [mem 0x000000007ff90000-0x000000007ff9dfff] ACPI data
[... cut ...] bsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; r 0
[&nbsp; &nbsp; 1.605806] brd: module loaded
[&nbsp; &nbsp; 1.605836] loop: module loaded
[&nbsp; &nbsp; 1.611789] usbcore: registered new interface driver rtsx_usb
[&nbsp; &nbsp; 1.611813] Loading iSCSI transport class v2.0-870.
[&nbsp; &nbsp; 1.611921] iscsi: registered transport (tcp)
[&nbsp; &nbsp; 1.611922] megasas: 07.701.17.00-rc1
[&nbsp; &nbsp; 1.612171] ata_piix 0000:00:1f.1: version 2.13
[&nbsp; &nbsp; 1.614204] scsi host0: ata_p

gouchi, 2021-09-11:
Can you give also RA log with the nighty build ?

Before it was working for the device index, it was not disable with udev driver ?

Did you try also with `parport` or `linuxraw` controller driver ?

Thank you.

BDJoe360, 2021-09-11:
I always use only the Linux raw controller driver.
Nothing else can make psx controller appear.

The RA Log:
[INFO] [Config]: Loading default config.
[INFO] [Config]: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
[INFO] RetroArch 1.9.9 (Git 36888612eb)
[INFO] === Build =======================================
[INFO] CPU Model Name: Intel(R) Xeon(R) CPU           L3110  @ 3.00GHz
[INFO] 容量:  MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4
[INFO] Built: Sep 11 2021
[INFO] Version: 1.9.9
[INFO] Git: 36888612eb
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] libretro API 版本: 1
[INFO] 未按照 API 编译: 1
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [DRM]: Found 3 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 26 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Connector 2 connected: no
[INFO] [DRM]: Connector 2 has 0 modes.
[INFO] [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 1: (1920x1080) 1920 x 1080, 59.940201 Hz
[INFO] [DRM]: Mode 2: (1920x1080i) 1920 x 1080, 30.000000 Hz
[INFO] [DRM]: Mode 3: (1920x1080i) 1920 x 1080, 29.970100 Hz
[INFO] [DRM]: Mode 4: (1920x1080) 1920 x 1080, 50.000000 Hz
[INFO] [DRM]: Mode 5: (1920x1080i) 1920 x 1080, 25.000000 Hz
[INFO] [DRM]: Mode 6: (1920x1080) 1920 x 1080, 30.000000 Hz
[INFO] [DRM]: Mode 7: (1920x1080) 1920 x 1080, 29.970100 Hz
[INFO] [DRM]: Mode 8: (1920x1080) 1920 x 1080, 24.000000 Hz
[INFO] [DRM]: Mode 9: (1920x1080) 1920 x 1080, 23.976080 Hz
[INFO] [DRM]: Mode 10: (1600x900) 1600 x 900, 60.000000 Hz
[INFO] [DRM]: Mode 11: (1280x1024) 1280 x 1024, 60.019741 Hz
[INFO] [DRM]: Mode 12: (1280x720) 1280 x 720, 60.000000 Hz
[INFO] [DRM]: Mode 13: (1280x720) 1280 x 720, 59.940201 Hz
[INFO] [DRM]: Mode 14: (1280x720) 1280 x 720, 50.000000 Hz
[INFO] [DRM]: Mode 15: (1280x720) 1280 x 720, 30.000000 Hz
[INFO] [DRM]: Mode 16: (1280x720) 1280 x 720, 29.970100 Hz
[INFO] [DRM]: Mode 17: (1280x720) 1280 x 720, 24.000000 Hz
[INFO] [DRM]: Mode 18: (1280x720) 1280 x 720, 23.976162 Hz
[INFO] [DRM]: Mode 19: (1024x768) 1024 x 768, 60.003841 Hz
[INFO] [DRM]: Mode 20: (800x600) 800 x 600, 60.316540 Hz
[INFO] [DRM]: Mode 21: (720x576) 720 x 576, 50.000000 Hz
[INFO] [DRM]: Mode 22: (720x480) 720 x 480, 60.000000 Hz
[INFO] [DRM]: Mode 23: (720x480) 720 x 480, 59.940060 Hz
[INFO] [DRM]: Mode 24: (640x480) 640 x 480, 60.000000 Hz
[INFO] [DRM]: Mode 25: (640x480) 640 x 480, 59.940475 Hz
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
[INFO] [GL]: Found GL context: kms
[INFO] [GL]: Detecting screen resolution 1920x1080.
[INFO] [EGL] Found EGL_EXT_platform_base, trying eglGetPlatformDisplayEXT
[INFO] [EGL]: EGL version: 1.5
[INFO] [EGL]: Current context: 0x7f72bc007060.
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [GL]: Vendor: nouveau, Renderer: NV106.
[... 52 lines cut ...]
[WARN] [udev]: Mouse grab/ungrab feature unavailable.
[INFO] ALSA: Using signed 16-bit format.
[INFO] ALSA: Period size: 384 frames
[INFO] ALSA: Buffer size: 1536 frames
wifi is not available
[INFO] [Display]: Found display driver: "gl".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
function is no-op
[INFO] [SRAM]: 游戏存档不会被保存。
[INFO] [Playlist]: 正在读取历史文件: [/storage/.config/retroarch/content_history.lpl].
[INFO] [Playlist]: 正在读取历史文件: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] [Playlist]: 正在读取历史文件: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] [Playlist]: 正在读取历史文件: [/storage/.config/retroarch/content_image_history.lpl].
[INFO] [Playlist]: 正在加载收藏文件: [/storage/.config/retroarch/content_favorites.lpl].
function is no-op
function is no-op
function is no-op
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).

Thank you.

gouchi, 2021-09-12:
Hi,

Can you try to generate [joypad  autoconfig](http://www.lakka.tv/doc/Contributing-your-joypad-config/) because I don't see a lot of [config files for linuxraw](https://github.com/libretro/retroarch-joypad-autoconfig/tree/master/linuxraw).

Thank you.

BDJoe360, 2021-09-12:
Hello,
The joypads/Lirnuxaw folder is not generated in nightly-20210911.
To be exact, no files are generated in the joypads folder, which is completely empty.
In the previous version, a cfg file named psx controller was generated in joypads / lirnuxaw.
The contents are as follows:
input_a_btn = "1"
input_b_btn = "2"
input_device = "PSX controller"
input_down_axis = "+1"
input_driver = "linuxraw"
input_l2_btn = "4"
input_l_btn = "6"
input_left_axis = "-0"
input_r2_btn = "5"
input_r_btn = "7"
input_right_axis = "+0"
input_select_btn = "8"
input_start_btn = "9"
input_up_axis = "-1"
input_x_btn = "0"
input_y_btn = "3"

Thank you.

ToKe79, 2021-09-12:
@BDJoe360: are you talking about `/etc/retroarch-joypad-autoconfig/linuxraw` or the samba share `Joypads`? The first one must include these files:
```
./CHA2.cfg
./Arduino LLC Arduino Leonardo.cfg
./Microsoft X-Box 360 pad.cfg
./Sony Interactive Entertainment Wireless Controller.cfg
./Steam_Controller.cfg
./CHA1.cfg
./Toodles 2008 Cthulhu Mult.cfg
./Bliss_Box_4-Play.cfg
```
The same contents should be in `/tmp/joypads/linuxraw` (+ anything added to the samba share `Joypads/linuxraw` (which is `/storage/joypads/linuxraw`).

BDJoe360, 2021-09-14:
Hi,
I send the details of the question here:
https://forums.libretro.com/t/problem-description-of-gamecon-module/34920

Hope it can be solved!
Thank you.

ToKe79, 2021-09-14:
There is `Save controller profile` item - select it and it will create your configuration file in the `Joypads` folder.
[image]

BDJoe360, 2021-09-15:
HI,
Not so, as shown in the figure:

This will appear automatically after loading gamecon module.
Moreover, the joypads folder will automatically create the Linux raw folder and the configuration file of the psx controller.
Please help fix this problem.

Thankyou.

 

 
 

从QQ邮箱发来的超大附件

微信图片编辑_20210915101818.jpg (177.1K, 无限期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?k=313434645702cbcdb569b9731639041d040d5701085a0750485706555214020b5005195d565a551f03575756560b050404050100302636fcc7e4f1a98cff9a838588993b02090403550d05510109070a540c1a0e405e360f&t=exs_ftn_download&code=e44d0962

ToKe79, 2021-09-17:
So just to be clear:
- the Gamecon driver works - you can use the gamepad, assign buttons, play games
- however, when saving the config by `Save Controller Profile`, the config is not saved and you have to configure the gamepad every time you boot Lakka / restart RetroArch

Is this the case?

BDJoe360, 2021-09-17:
Hi,
1.Yes.
2.No, it's not like that.The config is saved.And No have to configure the gamepad every time you boot Lakka / restart RetroArch。As shown in my picture, the item of device index shows disabld instead of psx controller.Moreover, the controller profile config file will not be created in the joypads folder.That's the little problem.

Thankyou.

ToKe79, 2021-09-17:
And you are certain that the config file you have with the 2.x version of Lakka was created by RetroArch? Can you create folder `linuxraw` in the 3.x `joypads` and copy the config from 2.x to 3.x? Does that help in any matter?
Unfortunately I can't test this myself - I have no Gamecon adapter and even if I had, I don't have a computer with parallel port. I hope someone with the right set-up can help you.

BDJoe360, 2021-09-18:
Hi,
1.100% sure.
2.3. X does not create the file，Moreover, even if I copy the files created in the old version to 3. X, the disabled state of the device index cannot be changed.
3.In fact, I suspect that this problem may be caused by the lack of some parameters in the compilation process.

Thankyou.

gouchi, 2022-03-02:
@BDJoe360 Is still an issue with [nightly builds](https://nightly.builds.lakka.tv/) ?

Thank you.

Ntemis, 2022-06-10:
Stalled issue

commit ab378a22 (2021-09-10): Gamecon: enable additional PARPORT modules/features

fixes #1495
closed 2022-06-10 by Ntemis